frangoteam/FUXA #6: Load demo project from help menu

unocelli, 2019-03-21:


ATwins-8, 2019-11-26:
An error will be reported in fuxa-editor.min.js file: 
console.log :1 error: < Path > attribute d: expected move to path command ('m 'or' m '), "L 5 L 10 10 10";

unocelli, 2019-11-26:
Hi,
I know, I will try to fix it as soon as possible, however it has no
consequence on the correct functioning of the application.

ATwins-8, 2019-12-06:
When Firefox is running editing projects, the console will report an error: error typeerror: "u.clear is not a function". The reason for the error is the code in fuxa.editor.min.js. When you select Firefox to edit, drag the shape icon on the left to the canvas to report the above error. Not in Google.

ATwins-8, 2019-12-06:
Because you have packed the compressed code, I have not found the corresponding solution

closed 2020-05-23 by unocelli