frangoteam/FUXA #29: A bug when ungroup a shape

khajegan, 2019-07-07:
In editor page when i use a Proc.Eng shape like 'motor-simb', and after i click on ungroup option, the forcolor and backcolor of the shape automaticcaly sets to black.

unocelli, 2019-07-14:
Thanks khajegan,
I will fix it by disabling the ungroup button by selecting a ‘Controls’ or a ‘Proc. Eng.’ shape.

khajegan, 2019-07-16:
Dear unocelli
Please don't do that , i want ungroup capability for 'Proc. Eng' or any complex shape. please fix it.
Thanks

unocelli, 2019-07-16:
Dear khajegan,
would you explain to me what is your goal by trying to ungroup this shape?
These shapes are built as groups and have an attribute in order to be identified, if you try to ungroup and group again the shape loses the attribute (I can't distinguish that I'm groupping again a old shape).
The feature to creating your own shapes library is planned but later.

Regards

khajegan, 2019-07-21:
Dear unocelli
I want to use my own complex shape as svg file like your 'Proc. Eng' (i could do that by using your import function), and our users may want to change these shapes as they want or maybe they want to ungroup these shapes and then group some parts of them and create another complex shape.

unocelli, 2019-07-28:
Dear khajegan,
of course you will be able to ungroup and group again your complex shape, composed with a 'Proc. Eng' shape.
As explained you can not ungroup the only 'Proc. Eng' shapes alone...Let's discuss again after my modification.

Regards

khajegan, 2019-08-03:
Dear unocelli
Thanks , please tell me after you're done.
Please add another new options to add own complex shape as SVG, more easier than this.
Thanks a lot

ATwins-8, 2019-11-14:
Indeed, we all want to add our own SVG. I try to drag my own SVG from the mat expansion panel tag to the canvas, but it will fail. Do you want to change the function svgeditor. Addextension () in ext-bundle.min.js? Thank you for your reply!

unocelli, 2019-11-14:
Hi,
It's planned to create the possibility to import your own Shapes and Gauges. But I can't provide the exact date. I opend a new issue #41 and I will work with it soon.

unocelli, 2020-05-23:
Fixed:  ungroup a control or shape is prevented

closed 2020-05-23 by unocelli