frangoteam/FUXA #31: how to change the svgeditor' language

nulIptr, 2019-08-14:
I find some templete in client\src\app\editor\editor.component.html named svg_prefs,
but it don't work,and there's only 'client\src\assets\lib\svgeditor\fuxa-editor.min.js',I can't read it,may I have the source code?

unocelli, 2019-08-16:
Hi nullptr,
you're right Internationalization (i18n) is missing, I added the issue that I will try to implement as soon as possible

best regards

ATwins-8, 2019-11-16:
Hello! Unocelli. Why can't I find the SetColor function in this.winref.nativewindow.svgeditor during initialization? Console error: this.winref.nativewindow.svgeditor.setcolor is not a function? It seems very unstable

unocelli, 2019-11-16:
Hi,
where do you have this problem, in FUXA_master or in your application?
if it is in your application you have added fuxa-editor.min.js to your
start page 'index.html'
    <script src="assets/lib/svgeditor/fuxa-editor.min.js"></script>

ATwins-8, 2019-11-21:
The start page 'index. HTML' already exists, but if I add the dependency of node package in "scripts" in the. Angular-cli.json file, an error will be reported. However, during the first initialization, an error typeerror is reported: cannot read property 'SetColor' of undefined. Just execute this function for the second time.

unocelli, 2019-11-21:
Hi,
I know that if you add in angular-cli.json an error will be reported.
You can try anyway to add in your start page and tell me if it worked?
You have to add in index.html the following dependencies (like in
FUXA\client) which are necessary for the editor:

    
    <link rel="stylesheet"
href="assets/lib/svgeditor/jquery-plugin.min.css">

    <link rel="stylesheet"
href="assets/lib/svgeditor/extensions/ext-html_button.min.css">
    <link rel="stylesheet" href="assets/lib/svgeditor/svg-editor.min.css">
    <link rel="stylesheet" href="assets/lib/ext-styles.css">

    
    <script src="assets/lib/svgeditor/jquery.min.js"></script>
    <script
src="assets/lib/svgeditor/jquery-ui-1.8.17.custom.min.js"></script>
    <script src="assets/lib/svgeditor/jquery-plugin.min.js"></script>

    <script src="assets/lib/dygraphs/dygraph-combined.js"></script>
    <script src="assets/lib/dygraphs/dygraph-extra.js"></script>
    <script src="assets/lib/dygraphs/dygraph-synchronizer.js"></script>

    <script
src="assets/lib/svgeditor/extensions/ext-markers.min.js"></script>
    <script
src="assets/lib/svgeditor/extensions/ext-bundle.min.js"></script>

    <script src="assets/lib/svgeditor/fuxa-editor.min.js"></script>

    <script src="assets/lib/svg/svg.min.js"></script>
    <script src="assets/lib/svg/raphael.min.js"></script>

ATwins-8, 2019-11-26:
I've removed the dependency package in the angular-cli.json file and included all the import files you mentioned in the index.html file. However, ERROR TypeError: Cannot read property 'setColor' of undefined in the first load

unocelli, 2019-11-26:
Hi,
can you send me the source code?

ATwins-8, 2019-12-02:
`<meta charset="UTF-8">
    <meta name="viewport" content="width=device-width, initial-scale=1">
    
    <link rel="stylesheet" href="assets/lib/svgeditor/jquery-plugin.min.css">

    <link rel="stylesheet" href="assets/lib/svgeditor/extensions/ext-html_button.min.css">
    <link rel="stylesheet" href="assets/lib/svgeditor/svg-editor.min.css">
    <link rel="stylesheet" href="assets/lib/ext-styles.css">

    
    <script src="assets/lib/svgeditor/jquery.min.js"></script>
    <script src="assets/lib/svgeditor/jquery-ui-1.8.17.custom.min.js"></script>
    <script src="assets/lib/svgeditor/jquery-plugin.min.js"></script>

    <script src="assets/lib/dygraphs/dygraph-combined.js"></script>
    <script src="assets/lib/dygraphs/dygraph-extra.js"></script>
    <script src="assets/lib/dygraphs/dygraph-synchronizer.js"></script>

    <script src="assets/lib/svgeditor/extensions/ext-markers.min.js"></script>
    <script src="assets/lib/svgeditor/extensions/ext-bundle.min.js"></script>
    
    

    <script src="assets/lib/svgeditor/fuxa-editor.min.js"></script>

    <script src="assets/lib/svg/svg.min.js"></script>
    <script src="assets/lib/svg/raphael.min.js"></script>
    
    <script src="assets/lib/echarts.min.js"></script>
  </head>
  
  <body>
    <app-root>Loading...</app-root>
  </body>`

Console error：
core.js:1440 ERROR TypeError: Cannot read property 'setColor' of undefined
    at EditorComponent.setFillColor (editor.component.ts:500)
    at EditorComponent.onSelectView (editor.component.ts:798)
    at EditorComponent.loadHmi (editor.component.ts:243)
    at eval (editor.component.ts:121)
    at SafeSubscriber.schedulerFn [as _next] (core.js:4331)
    at SafeSubscriber.__tryOrUnsub (Subscriber.js:239)
    at SafeSubscriber.next (Subscriber.js:186)
    at Subscriber._next (Subscriber.js:127)
    at Subscriber.next (Subscriber.js:91)
    at EventEmitter.Subject.next (Subject.js:56)

The reported file error is in editor.component.ts
`    private setFillColor(event) {
        let color = event;
        if (color.charAt(0) === '#')
            color = color.slice(1);
        let alfa = 100;
        this.winRef.nativeWindow.svgEditor.setColor(color, alfa, "fill");
        // this.fillcolor;
    }`

unocelli, 2019-12-03:
Hi,
I think is a synchronization problem: the callback from server that enable
the frontend to load the project (loadHmi) and the svgEditor that it's not
ready yet. I will try to reproduce in my PC and to fix it.

ATwins-8, 2019-12-05:
Thank you! If you find a solution, please let me know. I am also trying to solve this problem.

closed 2020-09-04 by unocelli