frangoteam/FUXA #32: Add support for InfluxDB

tisbris, 2019-08-15:
This project looks very good...

I would suggest that the historic data, incl. Alarms & Events are stored inInfluxDB, because this database is fast and dashboard tools are already available.

Kortenbach, 2022-06-14:
Good idea! Once the data is in InfluxDB you can view it in Grafana or some other tool. It would be a tremendous enhancement!

superkalt, 2022-06-14:
yes i love the tick stack! grafana to!

Kortenbach, 2022-06-15:
Afaik, FUXA has no long therm data storage. InfluxDB provides an easy way to add long therm storage. Once added you can view the data using a tool like Grafana.

unocelli, 2022-09-08:
going to close this as resolved. let me know if you have any issues.

closed 2022-09-16 by unocelli

henjoe, 2022-11-05:
I hope there are still support for InfluxDB 1.8 version. 
The supported is only for V2.

[image]

Kortenbach, 2022-11-05:
Great work!