frangoteam/FUXA #33: App not loading on Google Cloud

tisbris, 2019-08-16:
I have trouble getting the app to load on Google Cloud (Ubuntu 18.04).

I have NGINX on port 80, which redirect to port 1881 

location / {
        proxy_pass http://localhost:1881;
        proxy_http_version 1.1;
        proxy_set_header Upgrade $http_upgrade;
        proxy_set_header Connection 'upgrade';
        proxy_set_header Host $host;
        proxy_cache_bypass $http_upgrade;

The url 'example.com' is loading FUXA web server, because header is changed to FUXA and the page shows 'loading... ', which is what is found in /client/dist/index.html, but how do I get the app to load all the good stuff?

I can also see that client are connected to server and no errors are recorded in FUXA logs.

I would like to test this project, so any help is appriciated.

tisbris, 2019-08-16:
The problem seems to be related to NGINX proxy_pass.

References are not working behind this proxy.

I have changed root to /home/<user>/fuxa/client/dist/, which makes some of the app loading, but with many errors.

Please advise how to load this app using NGINX proxy.

unocelli, 2019-08-17:
Hi tisbris,
what does the browser (chrome) console display?

tisbris, 2019-08-18:
this is from Google console:

`svg-editor.min.js:1 Failed to load resource: the server responded with a status of 404 (Not Found)
svg-editor.min.js:1 Failed to load resource: the server responded with a status of 404 (Not Found)
vendor.bundle.js:168563 Could not find HammerJS. Certain Angular Material components may not work correctly.
webpackJsonp.../../../material/esm5/core.es5.js.MatCommonModule._checkHammerIsAvailable @ vendor.bundle.js:168563
vendor.bundle.js:136848 Angular is running in the development mode. Call enableProdMode() to enable the production mode.
HomeComponent.html:12 ERROR Error: ExpressionChangedAfterItHasBeenCheckedError: Expression has changed after it was checked. Previous value: 'false'. Current value: 'true'.
    at viewDebugError (vendor.bundle.js:142928)
    at expressionChangedAfterItHasBeenCheckedError (vendor.bundle.js:142906)
    at checkBindingNoChanges (vendor.bundle.js:143075)
    at checkNoChangesNodeInline (vendor.bundle.js:147126)
    at checkNoChangesNode (vendor.bundle.js:147100)
    at debugCheckNoChangesNode (vendor.bundle.js:147929)
    at debugCheckDirectivesFn (vendor.bundle.js:147831)
    at Object.eval [as updateDirectives] (HomeComponent.html:12)
    at Object.debugUpdateDirectives [as updateDirectives] (vendor.bundle.js:147813)
    at checkNoChangesView (vendor.bundle.js:146938)
View_HomeComponent_0 @ HomeComponent.html:12
HomeComponent.html:12 ERROR CONTEXT DebugContext_
View_HomeComponent_0 @ HomeComponent.html:12
/api/settings:1 Failed to load resource: the server responded with a status of 404 (Not Found)
main.bundle.js:1591 load Project
main.bundle.js:1637 -save-
main.bundle.js:1577 HttpErrorResponse
(anonymous) @ main.bundle.js:1577
/socket.io/?EIO=3&transport=polling&t=MoZY6lO:1 Failed to load resource: the server responded with a status of 404 (Not Found)
/socket.io/?EIO=3&transport=polling&t=MoZY6_I:1 Failed to load resource: the server responded with a status of 404 (Not Found)
/socket.io/?EIO=3&transport=polling&t=MoZY7hL:1 Failed to load resource: the server responded with a status of 404 (Not Found)
/socket.io/?EIO=3&transport=polling&t=MoZY8NF:1 Failed to load resource: the server responded with a status of 404 (Not Found)
polyfills.bundle.js:5062 GET http://35.228.208.244/socket.io/?EIO=3&transport=polling&t=MoZY9cb 404 (Not Found)
scheduleTask @ polyfills.bundle.js:5062
webpackJsonp.../../../../zone.js/dist/zone.js.ZoneDelegate.scheduleTask @ polyfills.bundle.js:2500
onScheduleTask @ polyfills.bundle.js:2390
webpackJsonp.../../../../zone.js/dist/zone.js.ZoneDelegate.scheduleTask @ polyfills.bundle.js:2494
webpackJsonp.../../../../zone.js/dist/zone.js.Zone.scheduleTask @ polyfills.bundle.js:2325
webpackJsonp.../../../../zone.js/dist/zone.js.Zone.scheduleMacroTask @ polyfills.bundle.js:2348
scheduleMacroTaskWithCurrentZone @ polyfills.bundle.js:3207
(anonymous) @ polyfills.bundle.js:5094
proto.<computed> @ polyfills.bundle.js:3487
webpackJsonp.../../../../engine.io-client/lib/transports/polling-xhr.js.Request.create @ vendor.bundle.js:4004
Request @ vendor.bundle.js:3906
webpackJsonp.../../../../engine.io-client/lib/transports/polling-xhr.js.XHR.request @ vendor.bundle.js:3833
webpackJsonp.../../../../engine.io-client/lib/transports/polling-xhr.js.XHR.doPoll @ vendor.bundle.js:3863
webpackJsonp.../../../../engine.io-client/lib/transports/polling.js.Polling.poll @ vendor.bundle.js:4279
webpackJsonp.../../../../engine.io-client/lib/transports/polling.js.Polling.doOpen @ vendor.bundle.js:4224
webpackJsonp.../../../../engine.io-client/lib/transport.js.Transport.open @ vendor.bundle.js:3348
webpackJsonp.../../../../engine.io-client/lib/socket.js.Socket.open @ vendor.bundle.js:2760
Socket @ vendor.bundle.js:2633
Socket @ vendor.bundle.js:2540
webpackJsonp.../../../../socket.io-client/lib/manager.js.Manager.open.Manager.connect @ vendor.bundle.js:67225
(anonymous) @ vendor.bundle.js:67539
webpackJsonp.../../../../zone.js/dist/zone.js.ZoneDelegate.invokeTask @ polyfills.bundle.js:2514
onInvokeTask @ vendor.bundle.js:137897
webpackJsonp.../../../../zone.js/dist/zone.js.ZoneDelegate.invokeTask @ polyfills.bundle.js:2513
webpackJsonp.../../../../zone.js/dist/zone.js.Zone.runTask @ polyfills.bundle.js:2281
webpackJsonp.../../../../zone.js/dist/zone.js.ZoneTask.invokeTask @ polyfills.bundle.js:2589
ZoneTask.invoke @ polyfills.bundle.js:2578
timer @ polyfills.bundle.js:4147
polyfills.bundle.js:5062 GET http://35.228.208.244/socket.io/?EIO=3&transport=polling&t=MoZYArz 404 (Not Found)
scheduleTask @ polyfills.bundle.js:5062
webpackJsonp.../../../../zone.js/dist/zone.js.ZoneDelegate.scheduleTask @ polyfills.bundle.js:2500
onScheduleTask @ polyfills.bundle.js:2390
webpackJsonp.../../../../zone.js/dist/zone.js.ZoneDelegate.scheduleTask @ polyfills.bundle.js:2494
webpackJsonp.../../../../zone.js/dist/zone.js.Zone.scheduleTask @ polyfills.bundle.js:2325
webpackJsonp.../../../../zone.js/dist/zone.js.Zone.scheduleMacroTask @ polyfills.bundle.js:2348
scheduleMacroTaskWithCurrentZone @ polyfills.bundle.js:3207
(anonymous) @ polyfills.bundle.js:5094
proto.<computed> @ polyfills.bundle.js:3487
webpackJsonp.../../../../engine.io-client/lib/transports/polling-xhr.js.Request.create @ vendor.bundle.js:4004
[... 66 lines cut ...]
XMLHttpRequest.send (async)
scheduleTask @ polyfills.bundle.js:5062
webpackJsonp.../../../../zone.js/dist/zone.js.ZoneDelegate.scheduleTask @ polyfills.bundle.js:2500
onScheduleTask @ polyfills.bundle.js:2390
[... cut ...]
webpackJsonp.../../../../engine.io-client/lib/transports/polling.js.Polling.poll @ vendor.bundle.js:4279
webpackJsonp.../../../../engine.io-client/lib/transports/polling.js.Polling.doOpen @ vendor.bundle.js:4224
webpackJsonp.../../../../engine.io-client/lib/transport.js.Transport.open @ vendor.bundle.js:3348
webpackJsonp.../../../../engine.io-client/lib/socket.js.Socket.open @ vendor.bundle.js:2760
Socket @ vendor.bundle.js:2633
Socket @ vendor.bundle.js:2540
webpackJsonp.../../../../socket.io-client/lib/manager.js.Manager.open.Manager.connect @ vendor.bundle.js:67225
`
thx.

unocelli, 2019-08-18:
A questions: you are using NGINX as WebServer and only the client of FUXA project?

tisbris, 2019-08-19:
Nginx are used a webserver for all incomming requests.

Http request on port 80 are passed to Fuxa webserver on port 1881.

The issue seems to be how the html header are passed on to fuxa client from fuxa server when using Nginx.

unocelli, 2019-08-20:
I think you don't have to change root to /home/fuxa/client/dist/ because the directory contain the frontend resource which is served from the backend (FUXA webserver).
then the browser console display the same error message?

ATwins-8, 2019-11-09:
I have set the proxy.config.json file locally, and I have also got the database data normally. However, after packaging and publishing, I was unable to connect to the database. Hope to get help! Thank you!

unocelli, 2019-11-10:
Hi,
have you configured the uiHost in settings.js with the ip address (by
windows '...\AppData\Roaming\.fuxa' by linux '/home/.../.config/.fuxa')?

Best Regards
umberto nocelli

ATwins-8, 2019-11-12:
I didn't use the server, just the client, so I didn't involve settings.js. My server does not use node.js

ATwins-8, 2019-11-12:
If you have a detailed documentation, that's really great. Especially the view rendering after socket.io

unocelli, 2019-11-13:
Hi,
if you don't use the server i don't now how can i help you with the database.

I don't have detailed documentation...sorry. you can follow from TesterComponent to the GaugesManager (through HmiService) and to the different Gauges.

Frechete, 2025-06-12:
`map $http_upgrade $connection_upgrade {
    default upgrade;
    '' close;
}

server {
    listen 80 default_server;
    listen [::]:80 default_server;
    return 301 https://$host$request_uri;
    server_name _;
}

server {
    listen 443 ssl http2 default_server;
    listen [::]:443 ssl http2 default_server;
    server_name _;

    ssl_certificate     /etc/nginx/ssl/server.crt;
    ssl_certificate_key /etc/nginx/ssl/server.key;
    ssl_protocols       TLSv1.2 TLSv1.3;
    ssl_ecdh_curve      X25519:prime256v1:secp384r1;
    add_header Strict-Transport-Security "max-age=63072000" always;

    location / {
        proxy_pass http://127.0.0.1:1881;
        proxy_http_version 1.1;
        proxy_set_header Host $host;
        proxy_set_header X-Real-IP $remote_addr;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header X-Forwarded-Proto $scheme;
        proxy_set_header X-Forwarded-Host $host;
        proxy_set_header X-Forwarded-Port $server_port;
        proxy_buffering off;
        proxy_request_buffering off;
        client_max_body_size 100m;
        proxy_read_timeout 86400s;
    }
}` This is my setup, respberry pi; same problem, fuxa fails to load with error "server connection failed!" do you have a solution? or there is another way to have https for fuxa?

Frechete, 2025-06-17:
To answer my own question, fuxa would work behind a proxy if the setup is a production environment

closed 2025-07-24 by unocelli