frangoteam/FUXA #37: JSON data connectivity

kevin5210, 2019-11-03:
JSON data connectivity

unocelli, 2019-11-03:
Hi,
are you thinking a import export of devices and tags settings in JSON format?

aniston, 2020-11-20:
Hello @unocelli, lovely work 👏 , bravo, anzi bravissimo ;)
sad that i found your project so late, just got onto the installing and running it.

I guess @kevin5210 is pointing to the JSON data structures connectivity like having a JSON Datasource under Devices that can interact with simple JSON data for example coming in from an MQTT source or gateway or anything that can return JSON Objects

I have a couple of comments and ideas to the BACnet Device interface and especially the good old node-bacstack library from Fabio Huser @fh1ch which since about 2 years is not being actively developed, but I leave that for another thread.

unocelli, 2020-11-22:
Hi, grazie mille...sei benvenuto.
I find it a good idea, are you thinking of a dynamic import or a defined
JSON structure? You have an example to inspire me...

Grüsse
umberto

aniston, 2020-12-30:
@unocelli , sorry for this delayed reply, although over a month old, here is a sample JSON object output from the node-bacnet lib .
- in practice this could easily come from a MQTT Suscriber response, what definately will change is the Structure and what comes back as values.
- in this example I did a readPropopertiyMultiple of 3 properties [77,28,44,372] of a BACnet Device "Type 8" Object of an Instance "1010220".
- the object response structure is respectively the OBJECT_NAME, DESCRIPTION, FIRMWARE_REVISION, SERIAL_NUMBER this structure would be useful if it could be selected or filtered respectively based on the `values` keys, sadly the example is a bit confusing given the nested words from values and value.

```
{
  "len": 105,
  "values": [
    {
      "objectId": {
        "type": 8,
        "instance": 1010220
      },
      "values": [
        {
          "id": 77,
          "index": 4294967295,
          "value": [
            {
              "value": "G003_S01_AS01",
              "type": 7,
              "encoding": 0
            }
          ]
        },
        {
          "id": 28,
          "index": 4294967295,
          "value": [
            {
              "value": "DDC Station DOI 247 HLK Gebäude 3",
              "type": 7,
              "encoding": 0
            }
          ]
        },
        {
          "id": 44,
          "index": 4294967295,
          "value": [
            {
              "value": "V3.1.5b890",
              "type": 7,
              "encoding": 0
            }
          ]
        },
        {
          "id": 372,
          "index": 4294967295,
          "value": [
            {
              "value": "000001493079",
              "type": 7,
              "encoding": 0
            }
          ]
        }
      ]
    }
  ]
}
```

saluti,
Aniston.

unocelli, 2021-01-01:
@aniston Currently I'm working on something like 'http request' node by node-red, where I can receive data streams from Web API in different formats (CSV, JSON, ...)

saluti

concept10, 2021-01-01:
You're doing great work here @unocelli.  Your project combines well with node-red for consuming data.

unocelli, 2021-11-16:
This feature was implemented. I will therefore close this issue, feel free to re-open it if something is still not working.

closed 2021-11-16 by unocelli