frangoteam/FUXA #43: Is it possible to integrate to netcore backend ?

masterchen, 2019-11-26:
Is it possible to integrate to netcore backend ? if so please give some advice how to , thanks.

unocelli, 2019-11-29:
Hi,
with asp.net core, I find it a good idea, the FUXA project already has its backend (node.js), a new repository should be created...then it would be better to create modules for the frontend: fuxa-editor to create the views and fuxa-view to show them

masterchen, 2019-11-29:
Thanks! Do you have plan to give an example ?

unocelli, 2019-12-13:
Hi,
I can try but next year...it is not so far.

q383186554, 2020-06-23:
HI,
have you start the project with asp.net core,dear author,I am excepting for it ,I really neet it.

unocelli, 2020-06-28:
Hi,
yes i started, as soon as i have defined the structure i want to publish it so that maybe someone helps me.

q383186554, 2020-07-03:
hi，
ok，i want to help, but it is really hard for me to build the hole program,i can only use or change some of it,If the backend program is ready i will use it and debug it

unocelli, 2022-10-09:
I'm going to close this due to inactivity.

closed 2022-10-09 by unocelli

w8w8w8, 2023-06-04:
hi, use .net6 or 7 is good idea in server. hope !!!

luy710, 2023-11-25:
> I'm going to close this due to inactivity.

Where is the .NET Core version? Can I participate in it?