frangoteam/FUXA #44: fuxa-editor.js source code?

xujingzhou, 2019-11-26:
I want to see fuxa-editor.js source code, not fuxa-editor.min.js in client. Please tell me where I can find it.  Thanks!

unocelli, 2019-11-26:
Hi,
I have not yet published the code of 'fuxa-editor.min.js' on github because
I would like to make a refactoring of this, and prefer to do it as soon as
it is finished (in any case the code is based on svg-edit version 2.8.1)

best regards

xujingzhou, 2019-11-28:
hi，
I want to extend several new graphic primitives on the edit panel, so I need the fuxa-editor.js source. Can you send the source code of fuxa-editor.js to my email (xujingzhou2016@gmail.com)?  Thank you very much!

xujingzhou, 2019-11-29:
sorry，should be fuxa-editor.ts。thx

closed 2019-12-06 by xujingzhou