frangoteam/FUXA #47: How does the "Align Right" button call the function

XiaoTongTang, 2020-01-17:
How does the "Align Right" button call the function?You don't use the "Onclick" property, nor is it a routing feature of the angular framework
[image]
[image]

unocelli, 2020-01-17:
Hi, no it isn't an angular component, in fact, it is a normal javascript ID
binding  made through fuxa-editor.min.js. The editor is a modified version
of SVG-edit (https://github.com/SVG-Edit/ SVG-edit).

XiaoTongTang, 2020-01-21:
Thank you very much for your help, in that case, I will go to check out the project you provided called "svg_edit"

closed 2020-01-21 by XiaoTongTang