frangoteam/FUXA #48: Extend svg editor with new shapes

albosmart-ro, 2020-02-05:
how do i extend the svg editor to append new predefined shapes?

```
<div class="svg-tool-button" [ngClass]="{'svg-tool-active': isModeActive('custom-shape')}" (click)="setMode('custom-shape')">
              <span class="icon-tool icon-custom-shape"></span>
            </div>
```

unocelli, 2020-02-07:
Hi,
I'm currently working on it, my goal would be to make editable
and extendable default shapes and with the possibility of setting events to
them.

unocelli, 2020-03-06:
Hi,
I just added the feature to extend shapes and to animate them 

Kind regards

unocelli, 2020-03-22:
I integrated the following gauge [https://bernii.github.io/gauge.js/](https://bernii.github.io/gauge.js/)

closed 2020-03-22 by unocelli