frangoteam/FUXA #51: Add new shape manually to editor

vuducminh, 2020-03-24:
How do I add/replace the shape in editor? For example, I want to replace compressor-fan.svg to other svg of my own. Thanks

unocelli, 2020-03-24:
Hi,
I added [HowTo define Shapes](https://github.com/frangoteam/FUXA/wiki/HowTo-define-Shapes) in Wiki, I hope it helps you.

vuducminh, 2020-03-25:
thanks. I clone your master but why I cannot add text when in editor mode?. It is black and I cannot text anything inside. I change to tag v1.0.3 and text works ok.

unocelli, 2020-03-25:
Hi,  If you would like to add a static text (label) you have to use the
toolbox ‘T’ in ‘General’, who had an error that I fixed and pushed. The
textbox is used to set device value, in this control you can not define a
text content in editmode. If by addition the textbox field is shown in
black you can change the color how you like (sometimes it happens and I
will try to fix it).
thanks

unocelli, 2020-03-25:
To define your own shapes you have to use the new pushed code (v1.0.4) that I haven't released yet.

closed 2020-04-15 by kotori94