frangoteam/FUXA #54: Read Memory Data of PLC S7

kotori94, 2020-04-01:
Hi, I can read Data Block of PLC S7 (e.g: DB1.DBD0), but I cannot read Memory Data of PLC S7 (e.g: MD44). How can I read that? thanks

unocelli, 2020-04-01:
Hi, the node-snap7 should support it but is missing in the wrapper that I have implemented.
I don't use read Memory. I can try to extend my wrapper...How can I simulate PLC Memory Data in my PC?

kotori94, 2020-04-02:
Hi, node-snap 7 support S7Client.MBRead and MBWrite function, but your source code only implemented DBRead and write. About simulating, you can try s7 simulator software. Hope you can fix this soon.

kotori94, 2020-04-02:
Or maybe you should try ReadMultiVars and WriteMultiVars function from node-snap7 instead of DBRead only

unocelli, 2020-04-04:
Hi, I agree that it is better to use ReadMultiVars/WriteMultiVars for all other kinds of variables.
I will try to fix it. thanks

kotori94, 2020-04-10:
Hi, how was it going? I tried to custom code but still cannot read it

unocelli, 2020-04-10:
Hi, I'm working to fix it. I will use ReadMultiVars / WriteMultiVArs only
for Output, Input and Merkers, the DB have to be grouped and operated with
DBRead for performance aspects by DAQ.
To simulate the PLC Server I'm using the CSharp Project from
http://snap7.sourceforge.net/

Hi, how was it going? I tried to custom code but still cannot read it

unocelli, 2020-04-13:
Hi, I fixed it. Thanks

kotori94, 2020-04-15:
It worked. Thank you

closed 2020-04-15 by kotori94