frangoteam/FUXA #56: Table of data feature

kotori94, 2020-04-15:
Hi, besides chart, I want to see table of data. For example, I want to see value of property each 30 minutes in last 3 days (can filter and choose value to add as well).
Hope you can help me. Thanks.
[image]

unocelli, 2020-04-15:
Hi, I can add a function to export the chart data in CSV format in order to analyze in excel.

kotori94, 2020-04-16:
Thanks. But How about the table in Scada HMI, like chart?. If it is to analyze, I can read sqlite db and analyze it later. A table in HMI can help me quickly see what is happening recent hours in the screen

unocelli, 2020-04-16:
Personally, I prefer to look at a chart than a table, in any case I have also planned to implement the table or list controls, there is in my to do list. Thanks.

kotori94, 2020-04-17:
Great! Hope it would be done soon.

kotori94, 2020-04-21:
Hi, could you give me the source code of fuxa-editor.min.js, so I can add the new module likes tables by myself?. Or you can guide me how to do it. Thanks

unocelli, 2020-04-21:
Hi, You don't need to change fuxa-editor.min.js, you can make it by add a new shape type with 'foreignObject' as type and then add new components (like for example table-property and ngx-table) in angular frontend. if you want I can try to guide you?

kotori94, 2020-04-22:
Great. I really want to implement table component to see data on HMI. Could you make a guide about this? Thanks

unocelli, 2020-04-26:
Hi, I created a 'ctrl-table' branch

destiny456, 2021-05-11:
Hi,export the chart in csv format,Did you add this function？

peakcontrol, 2021-05-16:
Hello,Do you have plans to merge Merge this branch  into master?

unocelli, 2021-05-17:
Hi,
the branch is not current and doesn't contain the feature. I’m working on a cards view where you can define the content of the card as table, list, etc….it's coming soon.

peakcontrol, 2021-05-19:
Looking forward it：）

anand1606, 2021-11-17:
still waiting table component and reporting tools, any update ?

unocelli, 2021-11-17:
Hi,
not yet, is always on the todo list...it would be better if you open another issue for the reporting tools and explain what you mean.

unocelli, 2022-07-29:
going to close this as resolved. let me know if you have any issues.

closed 2022-07-29 by unocelli