frangoteam/FUXA #61: OPCUA Variant(Scalar<ExtensionObject>

alenck, 2020-05-27:
extension of the data value type to objects.

unocelli, 2020-10-15:
Sorry I'm late, 
I don't know this type of object...can I bind it to svg object or control like a primitive type (integer, float, boolean, ...)?

alenck, 2020-12-26:
[image]

So now i see it but as [object Object] need to unpack it to primitive type(or have option if object additional select properties)

Possible?

unocelli, 2020-12-26:
To test and investigate I need a OPCUA Server with ExtensionObject. 
There are in the Demo Server from [unified-automation](https://www.unified-automation.com/downloads/opc-ua-servers.html)?