frangoteam/FUXA #62: Modbus TCP/RTU

Harshid93, 2020-06-01:
I need modbus Protocol in Fuxa SCADA.
Can you please suggest something to me.

unocelli, 2020-06-02:
Hi,
The Modbus TCP/RTU is now supported v1.0.5

Harshid93, 2020-06-03:
Thanks for your Reply sir.
It's Really appriciating..
Can yiou please let me knoe From where can i download v1.0.5? can you paste link here , i am unable to find out.

Harshid93, 2020-06-03:
Thanks for your reply
i tried this verson. v1.0.5.

While i am installting this through CMD window so manny errors are ther in npm Install command.
any system requirement or specifica node js & npm verson i have to use for this?
V 1.0.4 is working good with my system

unocelli, 2020-06-03:
Hi, try to remove by server the package-lock.json file, delete the ./node_modules and install again

Harshid93, 2020-06-03:
Thanks It's Working Now...!

unocelli, 2020-06-03:
Ok, I have to remove the file from the current version. Thanks

Harshid93, 2020-06-04:
In Modbus TCP Lib.
I can only make Single TAG in Tag list.
If i am making two holding Register i will stop communication with Modbus Server.

In Command Prompt this will apperas.

..................................................

(node:5456) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 57)
{ data: [ 123 ], buffer: <Buffer 00 7b> }
{ data: [ 56, 0 ], buffer: <Buffer 00 38 00 00> }
(node:5456) UnhandledPromiseRejectionWarning: RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds
    at boundsError (internal/buffer.js:75:11)
    at Buffer.readInt16BE (internal/buffer.js:497:5)
    at Object.parser (C:\Users\BOB\Desktop\FUXA-1.0.5\FUXA-1.0.5\server\runtime\devices\modbus\datatypes.js:5:61)
    at C:\Users\BOB\Desktop\FUXA-1.0.5\FUXA-1.0.5\server\runtime\devices\modbus\index.js:345:59
    at Array.map (<anonymous>)
    at C:\Users\BOB\Desktop\FUXA-1.0.5\FUXA-1.0.5\server\runtime\devices\modbus\index.js:343:30
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
(node:5456) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 58)

..........................................................................................

unocelli, 2020-06-04:
Hi, I fixed it. Thanks

Harshid93, 2020-06-05:
Ok it's Seem Working Now..
I am doing some more testing i will let you know if help i required from your side.

thanks...!

Harshid93, 2020-06-06:
Can We Add Allen Bredley Driver also here?
Rockwell PLC?

unocelli, 2020-06-09:
Hi, I don't know this plc and i don't know if there are drivers for javascript.
i found [https://github.com/cmseaton42/node-ethernet-ip](https://github.com/cmseaton42/node-ethernet-ip) ...maybe it works...how can we test it?

concept10, 2020-07-27:
Very impressed with this project.  Do you plan to release or publish any documentation regarding the writing and testing of device driver components?  I could help with the documentation.

unocelli, 2021-02-23:
@concept10 I'm sorry for the delay or I don't know if I've already reply you, anyway if your offer of help with the documentation is still valid you are welcome...you can contact me (4frango@gmail.com)

closed 2021-02-23 by unocelli