frangoteam/FUXA #64: USe this With Raspberry Pi

Harshid93, 2020-06-06:
I want to use this scada in Raspberry pi.
How can i use it?

unocelli, 2020-06-07:
Hi, in my raspi 3B work fine, a little warning by installation

python v2.7.13
nodejs v10.16.3
npm v6.9.0

 I tested it with modbus and opcua

closed 2020-09-29 by unocelli

Harshid93, 2020-12-20:
csn you please provide with latest versions?

unocelli, 2020-12-20:
what do you mean?

Harshid93, 2020-12-20:
I only Want to Know , IF we want to run this over Latest Version of python 
nodejs npm is it possible?
Coz Every time i need to Downgrade My Software Version to Run this...

concept10, 2020-12-20:
> I only Want to Know , IF we want to run this over Latest Version of python
> nodejs npm is it possible?
> Coz Every time i need to Downgrade My Software Version to Run this...

You should be able to run this software on any platform that supports the requirements, including raspberry pi.  You would be better served posting the versions of the software that you are currently attempting to run this on (for example node -v && npm -v).