frangoteam/FUXA #65: Setting Value of a variable

pedromartins200, 2020-06-17:
Currently in fuxa there is the possibility to add an event (onclick) to set value of a variable. 
In the creation of that event you need to set a predefined value.

This should be different. The user should be able to set value to any value.

Example: You click the shape that has that event, it should popup a modal containing an input that allows us to set any value that we want.

Thank you.

unocelli, 2020-06-23:
Hi,
I think it's best to be able to define a dialog, so that we can define all the desired inputs.
I think with the last commit this feature is possible, look by events 'Set Input'. The event to close the dialog is still missing.

unocelli, 2020-07-31:
The event to close the dialog is already implemented.

closed 2020-07-31 by unocelli