frangoteam/FUXA #66: FUXA doesn't cluster with PM2 (Process Manager)

concept10, 2020-08-03:
I generally use PM2 with nodejs projects to manage restart, load balancing, and the easy nodejs clustering feature.

FUXA doesn't simply run when using the command pm2 start **-i max** ~/someDirectory/fuxa/server/main.js --name FUXA 

Still investigating...

unocelli, 2020-09-29:
Hi, sorry for the delay,
I don't have much practice with PM2...you have some news?

concept10, 2020-12-20:
@unocelli This issue is still present, however, I believe your application may not be architected for clustering.  Have you any experience with the nodejs cluster module?  Okay, if not.  I would like to work on this with you.

unocelli, 2020-12-20:
No, I don't have experience with nodejs cluster module. Of course, you're welcome, I would very much appreciate a collaboration with you.

OptifySudarshanPatil, 2022-12-14:
fuxa can be indirectly  started with pm2
What i did?

1. I launched node-red with pm2
2. I created **.BAT** file with only 'fuxa' text
3. called above mention BAT file from Node-red "**EXEC**"
4. voila!