frangoteam/FUXA #68: Problem with Firefox by editing a project

unocelli, 2020-08-04:
By adding shapes to the views and changing between the views, the content of the views is lost

NikolayMurha, 2021-01-20:
Hello.
The problem in fuxa-editor.js, as I understood. 
By adding shape and click to current view on sidebar I have error: 

 TypeError: u.clear is not a function
    insertItemBefore fuxa-editor.min.js:15
    fixEnd fuxa-editor.min.js:18
    prepareSvg fuxa-editor.min.js:17
    setSvgString fuxa-editor.min.js:18
    setSvgString fuxa-editor.min.js:22
    loadView editor.component.ts:436
    onSelectView editor.component.ts:922
    View_EditorComponent_7 EditorComponent.html:25
    Angular 3
fuxa-editor.min.js:18:30471

After than I have POST request with `set-view` command and empty `items:[]`

​

unocelli, 2021-01-22:
Hi,
I don't know if the bug is always current, because I just tried after updating firefox, and I can't anymore reproduce it.

NikolayMurha, 2021-02-05:
> Hi,
> I don't know if the bug is always current, because I just tried after updating firefox, and I can't anymore reproduce it.
Cheers,
I have this bug on FF 86.0b6 (64-bit) and I can reproduce it stable.

In the working process on thee issue  #102 I used latest commit SVG-edit 2.8 https://github.com/SVG-Edit/svgedit/commit/396cce40ebfde03f7245c682041f63f07f69e3d3 branch as base for unminified [fuxa-editor.js](https://github.com/NikolayMurha/FUXA/blob/master/client/src/assets/lib/svgeditor/fuxa-editor.js) and I noticed that the error in FF is gone.

Problem with pathseg  polyfill. I found fix there https://github.com/SVG-Edit/svgedit/commit/44d1398c01df4cec5247ed0ea4b09a09388ddfab
and try to replace there https://github.com/NikolayMurha/FUXA/commit/9097e7de4f6ecf0871759a233619aad5d993ae02
The problem is no longer reproducible.

unocelli, 2021-02-06:
I'm sorry, you are right, I don't know what I tested :( .
Good job, Thanks

unocelli, 2021-02-08:
@NikolayMurha have you still encountered problems? I would close the issue.

NikolayMurha, 2021-02-09:
@unocelli no, I switched to FF after your commit, and haven't reproduced. I think bug fixed and you can close the issue.

closed 2021-02-09 by unocelli