frangoteam/FUXA #72: How to adapt abb plc equipment？

JasonLiu1021, 2020-08-25:
I found [node-ethernet-ip](https://github.com/cmseaton42/node-ethernet-ip) on github but I don’t know how to adapt it to FUXA.
 
[node-ethernet-ip](https://github.com/cmseaton42/node-ethernet-ip) can directly connect to ABB

unocelli, 2020-08-25:
Thanks for the suggestion, the package is very interesting. To integrate it you can look the other 3 supported protocols by 'server/runtime/devices'. I can try too but i don't have how to test it (I don't have the hardware). Is there any freeware emulator for pc?

JasonLiu1021, 2020-08-31:
> Thanks for the suggestion, the package is very interesting. To integrate it you can look the other 3 supported protocols by 'server/runtime/devices'. I can try too but i don't have how to test it (I don't have the hardware). Is there any freeware emulator for pc?

Thank you very much for your reply. I tried to search and found a software called[ RSLogix Emulator 5000](http://www.plcdev.com/a_quick_tutorial_on_rslogix_emulator_5000), but I don't know where to download and run it. I am still trying to find the download address.

JasonLiu1021, 2020-08-31:
> Thanks for the suggestion, the package is very interesting. To integrate it you can look the other 3 supported protocols by 'server/runtime/devices'. I can try too but i don't have how to test it (I don't have the hardware). Is there any freeware emulator for pc?

Found the download [address](magnet:?xt=urn:btih:A4AC6EAFF0E0E3A782A3F897F11B32082BA83F88), I haven't verified whether it can be used. Serial number: 2022123456

unocelli, 2020-09-01:
> Found the download address, I haven't verified whether it can be used.

Hi, Thanks
I found a video how to download and install it
https://www.youtube.com/watch?v=OAzf5QC2ZOM
I will try in the next day

Thanks for the suggestion, the package is very interesting. To integrate it

JasonLiu1021, 2020-09-03:
> Hi, Thanks I found a video how to download and install it https://www.youtube.com/watch?v=OAzf5QC2ZOM I will try in the next day Thanks for the suggestion, the package is very interesting. To integrate it
> […](#)

Thanks♪(･ω･)ﾉ   Looking forward to your adaptation .

unocelli, 2021-11-21:
Is there anyone who would like to help me by test the support of abb plc with the 'ethernetip' branch.

rvbatista, 2023-04-08:
ABB ac500 PLC series is able to communicate with Modbus tcp also.