frangoteam/FUXA #76: Displays for characters according to value

cbsbaosen, 2020-09-12:
I want to add a new control in Controls that displays the corresponding string based on the input value.It is similar to HTML_select, but does not require select, just display.At first I wanted to do this and submit the code, but looking at the code I found that I didn't have the extent-bundle.min.js source code, so I couldn't do it.Can you add this control?Thank you very much @unocelli 
[image]

unocelli, 2020-09-18:
Hi,
why don't you use the Output Control [image]

cbsbaosen, 2020-09-18:
I think the Output Control can only display the original value collected.Maybe I didn't make it clear.I want a control to do a simple mapping display, such as capture 1 to show "ONE" and 2 to show "TWO".I think the SELECT control has this capability, while the Output control does not. @unocelli

unocelli, 2020-09-18:
sorry ... you are right, i believe i can add a configurable attribute to the select control

cbsbaosen, 2020-09-18:
It doesn't matter.I think you could add a new control ,or add configuration to Output or select control, both of which would be good options. Thank you.
Finally, If you need me to improve the project together, I am willing to do my best.

unocelli, 2020-09-19:
I added the readonly option to the select control...
You are welcome

cbsbaosen, 2020-09-20:
Can it be displayed as a label instead of a selection box?Like the Output control.This makes the interface more user-friendly.

closed 2022-10-09 by unocelli