frangoteam/FUXA #77: Is 3D model integration possible in Fuxa?

drishyaraj24, 2020-10-15:
I am trying to integrate this three.js example to FUXA, but getting dependency issues. 
Please get back to me to solve this issue.
Thanks in advance
https://github.com/makimenko/angular-three-examples

Issues I am facing while integration
ERRORs in node_modules/three/src/core/BufferAttribute.d.ts(21,6): error TS1086: An accessor cannot be declared in an ambient context.
node_modules/three/src/core/InterleavedBufferAttribute.d.ts(19,6): error TS1086: An accessor cannot be declared in an ambient context.
node_modules/three/src/core/InterleavedBufferAttribute.d.ts(20,6): error TS1086: An accessor cannot be declared in an ambient context.
node_modules/three/src/renderers/WebGLRenderer.d.ts(35,31): error TS2304: Cannot find name 'OffscreenCanvas'.
node_modules/three/src/renderers/webgl/WebGLAttributes.d.ts(6,43): error TS2304: Cannot find name 'WebGL2RenderingContext'.
node_modules/three/src/renderers/webgl/WebGLUtils.d.ts(3,43): error TS2304: Cannot find name 'WebGL2RenderingContext'.
node_modules/three/src/renderers/webvr/WebVRManager.d.ts(13,15): error TS2304: Cannot find name 'VRDisplay'.
node_modules/three/src/renderers/webvr/WebVRManager.d.ts(14,21): error TS2304: Cannot find name 'VRDisplay'.

I tried to install all the missing modules listed in the issues,Then also same errors coming.
Please help me out with this
Type script Version : latest one

unocelli, 2020-10-20:
Hi,
I tried, I installed the 'three' version '0.88.0' of the example, it works

drishyaraj24, 2020-10-30:
Hi , could you please share the code , i tried to integrate ,but continously getting dependency issues.

drishyaanoop242019@gmail.com

unocelli, 2020-11-02:
I create the branch three-test

closed 2020-11-19 by unocelli