frangoteam/FUXA #78: Property window confirm buttons outside browser window

RG8550, 2020-10-30:
Can not confirm selected tags in Tag browser because button outside browser window.
[image]

For every object property window are the buttons outside browser window.
Can not scroll down to buttons.
[image]
[image]

Browser used: Chrome 86.0.4240.111

This is on a fresh install of ubuntu server 20.04.
nvm 0.34.0
node 10.17.0
npm 6.11.3
python 2.7.18

Nothing changed in settings.js.

No warnings/errors in fuxa.log
Only noticeable in api.log:
`"GET /assets/lib/gauge/gauge.js.map HTTP/1.1" 404 168 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/86.0.4240.111 Safari/537.36"`

unocelli, 2020-10-31:
Hi,
tanks for the feedback, I'll see if I can activate the scrollbar but I think to work comfortably with a graphics editor it is better to use a fairly large display and you know you can easily zoom with browser (CTRL +/-)

RG8550, 2020-11-02:
Thanks for looking to get scrollbar working.

Yes buttons are visible when zooming out browser.
But browser zoom does not work when in views editor, need to go to different page to zoom out.

unocelli, 2021-05-07:
I'm going to close this as resolved. let me know if you have any issues.

closed 2021-05-07 by unocelli