frangoteam/FUXA #81: Unable to install

xavierallem, 2020-12-04:
[image]

getting this error any solution?
 nodejs and npm are installed to the latest

unocelli, 2020-12-04:
Hi,
are you sure of the nodejs version? I see in error output node v4.2.6

xavierallem, 2020-12-06:
@unocelli Thank you for your reply, I have updated node.js to 15.3.0 

```bash
xavierallem@ubuntu:~$ nvm ls
->      v15.3.0
         system
default -> v15.3.0
node -> stable (-> v15.3.0) (default)
stable -> 15.3 (-> v15.3.0) (default)
iojs -> N/A (default)
unstable -> N/A (default)
lts/* -> lts/fermium (-> N/A)
lts/argon -> v4.9.1 (-> N/A)
lts/boron -> v6.17.1 (-> N/A)
lts/carbon -> v8.17.0 (-> N/A)
lts/dubnium -> v10.23.0 (-> N/A)
lts/erbium -> v12.20.0 (-> N/A)
lts/fermium -> v14.15.1 (-> N/A)
xavierallem@ubuntu:~$ 
```

I used the following command `npm install` in server folder
```bash
xavierallem@ubuntu:~/Desktop/Internship/FUXA-master/server$ npm install

up to date, audited 475 packages in 3s

28 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
```

got this, but when I run `npm start` i get this following error

[image]

Attaching the log file below

[2020-12-06T19_47_11_219Z-debug.log](https://github.com/frangoteam/FUXA/files/5649443/2020-12-06T19_47_11_219Z-debug.log)

xavierallem, 2020-12-06:
@unocelli 

I updated npm again with a different method and here is the log error attached below

[2020-12-06T20_00_31_458Z-debug.log](https://github.com/frangoteam/FUXA/files/5649451/2020-12-06T20_00_31_458Z-debug.log)

unocelli, 2020-12-06:
Hi,
currently the new Node version are not supported, you need to have installed Node (Version 10 or 12) and NPM (Version >=6.11), as it is explained in **Installing and Running** section

unocelli, 2020-12-17:
Closing this due to inactivity. Please respond if this issue should be reopened.

closed 2020-12-17 by unocelli