frangoteam/FUXA #82: Flow of server

vipulmistry1792, 2020-12-05:
Can i know the flow of the server architecture and also i want to integrate with aws

unocelli, 2020-12-06:
Hi,
interesting, I hope this little diagram can help you...I would like to know in a little more detail what is your goal?

[image]

vipulmistry1792, 2020-12-06:
Dear sir,
Thank you for your reply i just want to know that am just trying to connect
siemens plc for iot project but i can not connect so i found your code in
github i want to trying to understand how your code work so i can make
logic for my project. Do you have sample project for siemens protocol that
you use in your project?

unocelli, 2020-12-06:
to connect siemens plc I use:
- [node-snap7](https://github.com/mathiask88/node-snap7)
- [node-opcua](https://github.com/node-opcua/node-opcua) for family 1500

both have good sample

unocelli, 2020-12-17:
Closing this due to inactivity. Please respond if this issue should be reopened.

closed 2020-12-17 by unocelli