frangoteam/FUXA #83: strange behaviour on svg import

nkn134, 2020-12-18:
Drag and drop an image (or an svg file) actually works, but imported element cannot be moved until it's ungrouped/grouped some times.
Chrome console return that 
[image]

Latest Google Chrome version
Latest Fuxa version
Ubuntu 20.04

unocelli, 2020-12-18:
Hi,
can you send me one of these pictures?

nkn134, 2020-12-18:
Happens with every image, anyway that's one of them
I can't upload svg files, so i renamed in png
That doesn't seem to be correlated to file format, but somewhat random
[image]

pedromartins200, 2021-02-24:
I had similar situation.
Add the following to your svg

xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" xmlns:svg="http://www.w3.org/2000/svg" xmlns:dc="http://purl.org/dc/elements/1.1/" xmlns:cc="http://web.resource.org/cc/" xmlns:rdf="http://www.w3.org/1999/02/22-rdf-syntax-ns#" version="1.1"

unocelli, 2021-05-07:
I'm going to close this as resolved. let me know if you have any issues.

closed 2021-05-07 by unocelli