frangoteam/FUXA #85: Exception thrown on fresh install

IngoDe, 2020-12-20:
Hi frangoteam, Umberto,

on a fresh install of FUXA with "npm start" under /server the following exception is thrown (reproducible, V1.0.7, V1.0.8):
`-------------------------------------------
> node main.js

internal/fs/utils.js:759
    throw new ERR_INVALID_ARG_TYPE('mode', 'integer', mode);
    

TypeError [ERR_INVALID_ARG_TYPE]: The "mode" argument must be integer. Received type function ([Function (anonymous)])
    at Object.copyFileSync (fs.js:1989:10)
    at Object.<anonymous> (C:\Users\XXXXX\Programme\Fuxa\201219\fuxa\server\main.js:39:8)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
    at Module.load (internal/modules/cjs/loader.js:928:32)
    at Function.Module._load (internal/modules/cjs/loader.js:769:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:72:12)
    at internal/main/run_main_module.js:17:47 {
  code: 'ERR_INVALID_ARG_TYPE'
}
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! fuxa-server@1.0.8-beta start: `node main.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the fuxa-server@1.0.8-beta start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\XXXXX\AppData\Roaming\npm-cache\_logs\2020-12-20T11_42_57_370Z-debug.logfs.copyFileSync
-------------------------------------'

Responsible are the lines 30..44 in "main.js":  the user-settings under " _appdata/settings.js" are not found, routine
executes the "else" condition and crashes in "fs.copyFileSync(...)".

Actual workaround for me: a 0-Byte file "settings.js" under "_appdata/"

Best regards 
Ingo

unocelli, 2020-12-20:
> at Object. (C:\Users\XXXXX\Programme\Fuxa\201219\fuxa\server\main.js:39:8)

Hi,
I think is a synchronization problem by start the first time...
try it:
1° remove _appdata folder
2° start server
3° stop server
4° start server again

IngoDe, 2020-12-20:
Hi Umberto,
thanks for immediate reply. Not working under V1.0.7 and V1.0.8
Regards 
Ingo

unocelli, 2020-12-20:
nodejs and npm version?

IngoDe, 2020-12-20:
npm : 6.14.9
node: 14.15.2
Win 10: 2004

Attached you find the produced logfile.

unocelli, 2020-12-20:
Ok,
how I describe in installation section you have to use nodejs version 12 or
10 currently v14 is not supported
regards
umberto

IngoDe, 2020-12-20:
Hi Umberto,
sorry my mistake...

IngoDe, 2020-12-20:
The following patch solves this issue for me but it is sure that  i may run into further difficulties and i will not stress this
community with further issues. EOL of nodejs v12 is too short for us even in testing environment.

Solution for nodejs V14:
In main.js line 40 change 
fs.copyFileSync(defaultSettings, userSettingsFile, (err) => {
to
fs.copyFileSync(defaultSettings, userSettingsFile, COPYFILE_EXCL, (err) => {
In the head of this file add:
const { COPYFILE_EXCL } = fs.constants;

Umberto: thanks for your support!! Can i close this issue ?
Regards 
Ingo

unocelli, 2020-12-20:
Hi Ingo,
Yes you can close this issue.
I will try to fix for the support of nodejs V14, thanks for your tip.

Regards
umberto

unocelli, 2020-12-28:
Fixed, thanks

closed 2020-12-28 by unocelli

malcore10, 2022-12-05:
try with nodejs 14.17.4 (includes npm 6.14.14) [wiki](https://github.com/frangoteam/FUXA/wiki/Installing-and-Running)