frangoteam/FUXA #86: Issue In Raspberry PI

Harshid93, 2020-12-22:
Hello ,
we are trying to Install this over Rasp berry pi. by lowering the NPM , Python and Node version..
But it's showing error then..
can i run it with Latest verison of all backend pythin , NPM etc.?
How can we update all this regularly?

unocelli, 2020-12-22:
Hi,
which version of node, npm, python you have installed?

unocelli, 2020-12-23:
Last installation test Raspberry Pi 3 B
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

```
$sudo apt-get update
$sudo apt-get remove nodejs npm
$curl -sL https://deb.nodesource.com/setup_12.x | sudo bash -
$sudo apt-get install nodejs 
$node -v
v12.20.0
$npm -v
6.14.8
$python --version
Python 2.7.16
```

unocelli, 2021-03-31:
I'm assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2021-03-31 by unocelli