frangoteam/FUXA #88: Connecting with another device

amirgol, 2021-01-03:
Hi,
I installed FUXA on raspberry OS (both Desktop and noob lite). In the desktop version I can run FUXA inside Chromium with "localhost:1881" but I can not connect to the FUXA from my laptop (for example: "192.168.1.50:1881"  ). How can I solve it?
Also how can I run it in the background? and Is it possible to run in in the boot time? I mean auto start after rebooting?
Best regards

unocelli, 2021-01-04:
Hi,
I can allow access outside the localhost by setting in 'FUXA/server/_appdata/settings.js' the 'uiHost' with the server ip address.
To start by booting I would use pm2: [PM2 startup](https://dev.to/bogdaaamn/run-your-nodejs-application-on-a-headless-raspberry-pi-4jnn)

lvta0909, 2021-01-31:
I struggled some time on a similar issue as I was trying to run Fuxa on docker.
Finally found the problem:

The settings.js comments is wrong/misleading:
```
   // By default, server accepts connections on all IPv4 interfaces.
    // To listen on all IPv6 addresses, set uiHost to "::",
    // The following property can be used to listen on a specific interface. For
    // example, the following would only allow connections from the local machine.
    //uiHost: "127.0.0.1",
```
Thats not true, as on main.js@268 it forces only localhost if uiHost is not configured.
settings.uiHost = settings.uiHost || 'localhost'; //"0.0.0.0";

Just change it to:
settings.uiHost = settings.uiHost || '0.0.0.0'; //"0.0.0.0";

Or under settings.js use:
    uiHost: "0.0.0.0",

unocelli, 2021-02-01:
Fixed, Thanks

closed 2021-03-31 by unocelli