frangoteam/FUXA #90: Missing #blur-filter mask on home page

NikolayMurha, 2021-01-19:
Hello.

Filter #blur-filter exists on editor but not on home page: 
https://github.com/frangoteam/FUXA/blob/master/client/src/app/editor/editor.component.ts#L430
As a result, gauge-semaphore don't work correctly.
Mac OS, Chrome.

unocelli, 2021-01-22:
Hi,
can you better explain what is not working correctly?
The Filter #blur-filter should be added to every view, in the home page too. Have you inspect the DOM elements (the first <svg> in <app-fuxa-view>)?

NikolayMurha, 2021-01-22:
When I add semaphore it's disappeared when gauges gauge condition is true, but should change color instead.
I investigated the problem and came to the conclusion that the filter element is missing.

Removing the line solves the problem: https://github.com/frangoteam/FUXA/blob/master/client/src/app/gauges/controls/gauge-semaphore/gauge-semaphore.component.ts#L53 
I think this is dirty solution, but I didn't figure out where the blur-filter gone.

[Screenshot 1](https://www.dropbox.com/s/zn78mmgs1xobm43/%D0%A1%D0%BD%D0%B8%D0%BC%D0%BE%D0%BA%20%D1%8D%D0%BA%D1%80%D0%B0%D0%BD%D0%B0%202021-01-22%20%D0%B2%2017.38.31.png?dl=0)
[Screenshot 2](https://www.dropbox.com/s/sj5eexg9tz5nmd5/%D0%A1%D0%BD%D0%B8%D0%BC%D0%BE%D0%BA%20%D1%8D%D0%BA%D1%80%D0%B0%D0%BD%D0%B0%202021-01-22%20%D0%B2%2017.39.12.png?dl=0)

unocelli, 2021-01-22:
OK, I have to investigate why by Mac OS, Chrome don't work. 
The question is also whether the blur effect is interesting?
[image]

NikolayMurha, 2021-01-22:
No, not interesting for me :)

unocelli, 2021-02-01:
I'm agree with you, the blur effect is not so sexy

NikolayMurha, 2021-02-02:
I would have leave it, but I could not figure out why this element missing on mac. Semaphore without blur better than transparent semaphore :)

closed 2021-04-05 by unocelli