frangoteam/FUXA #101: Gauge actions by more than one variable

NikolayMurha, 2021-02-03:
I need to display an element by two values. 
For example, when 'a=1 && b=0'

unocelli, 2021-02-04:
Hi,
I think that a good way to implement this feature is to give the possibility to define script.
Every view could be bind to one script, where you can define simple logic and variable to bind at gauges present in the view
The script will be in javascript, working in frontend. 
In my todo list I have also planned the scripts in the backend but for the logic of the processes

NikolayMurha, 2021-02-05:
Now I solve it on controller side. 
Return analog one variable by OPCUA with three posible values instead of boollean

unocelli, 2022-10-09:
Can be done using scripts, so I am closing this issue.

closed 2022-10-09 by unocelli