frangoteam/FUXA #109: How to integrate with external data source

gpism, 2021-02-23:
Hi Team ...thanks for this wonderful project ...i need to ingrate with our backend which sends data via websocket ...any pointers on how to do it

unocelli, 2021-02-23:
Hi, Thanks.
can you explain a little more in detail what is your goal?

gpism, 2021-02-24:
> Hi, Thanks.
> can you explain a little more in detail what is your goal?

we have a lot of devices that use MQTT  for which we have a WebSocket adapter  /eclipse paho and custom JSON  device message format ....I just want to update FUXA Webhmi UI from  MQTT/WebSocket message received

unocelli, 2021-02-24:
I am currently working on the integration of MQTT client (with MQTT.js) for subscription and publish function.
I will configure in FUXA frontend (host, topic,...) that in backend will communicate with the broker and not direct from frontend.

gpism, 2021-02-24:
> I am currently working on the integration of MQTT client (with MQTT.js) for subscription and publish function.
> I will configure in FUXA frontend (host, topic,...) that in backend will communicate with the broker and not direct from frontend.

ok can I contribute ? if there are development docs please let me know

unocelli, 2021-02-27:
you could test the version on mqttclient branch...eventually contact me on 4frango@gmail.com

gpism, 2021-03-22:
are there any steps to do so ...will test and report back

also for OPC UA ....which open source server do you recommend
https://github.com/eclipse/milo or
https://python-opcua.readthedocs.io/en/latest/server.html ?

unocelli, 2021-03-27:
The configurations of the MQTT client is not very different from the other devices connectivities...you have to define the broker address to connect.

About open source opc ua server i don't know them, but i tried to program a simple server with [node-opcua](https://github.com/node-opcua/node-opcua) and it work very well.

unocelli, 2021-08-08:
I am closing this due to inactivity.

closed 2021-08-08 by unocelli