frangoteam/FUXA #110: Webui designer not running

gpism, 2021-02-23:
Hi Team in latest build and demo link getting following error 
DevTools failed to load SourceMap: Could not load content for https://frangoteam.github.io/assets/lib/gauge/gauge.js.map: HTTP error: status code 404, net::ERR_HTTP_RESPONSE_CODE_FAILURE

Page is not loading...pl release a fix

unocelli, 2021-02-23:
Hi, Thanks...
I think I can fix the problem of the message
'DevTools failed to load SourceMap: Could not load content for https://frangoteam.github.io/assets/lib/gauge/gauge.js.map: HTTP error: status code 404, net::ERR_HTTP_RESPONSE_CODE_FAILURE'

but I don't understand if you encountered a problem by using the demo version https://frangoteam.github.io, what you mean with 'Page is not loading'?

gpism, 2021-02-24:
> Hi, Thanks...
> I think I can fix the problem of the message
> 'DevTools failed to load SourceMap: Could not load content for https://frangoteam.github.io/assets/lib/gauge/gauge.js.map: HTTP error: [...]

I meant to say  due to the error no UI is showing  on page ...this problem is on demo page as well as nodejs plugin

unocelli, 2021-02-24:
Please can you send me the platform description that are you using: 
 - OS version
 - npm version
 - nodejs version
 - browser version

gpism, 2021-02-24:
> Please can you send me the platform description that are you using:
> 
> * OS version
> * npm version
> * nodejs version
> * browser version

os : mac os  mojave , browser : chrome , npm : 6.14.10, node : v14.15.4

gpism, 2021-02-24:
i dont think demo and npm are working on any os as  I checked from ubuntu facing same issue in angularjs...kindly fix asap if possible ...i want to show case fuxa in our office ...thanks

unocelli, 2021-02-24:
I will try to help you but I'm a little confusing, I can even see the demo [https://frangoteam.github.io](https://frangoteam.github.io) from my smartphone...can you send me a screenshot of your browser and console with error message?

gpism, 2021-02-24:
> I will try to help you but I'm a little confusing, I can even see the demo https://frangoteam.github.io from my smartphone...can you send me a screenshot of your browser and console with error message?
> The demo is only a frontend application to show how the editor work.
I tried on  Firefox , Opera and Chrome same error on all , maybe you have cached version 
https://share.getcloudapp.com/Qwu9dEpq
[image]

unocelli, 2021-02-24:
I update the demo version...you should have another error but maybe you see the blue button
[image]

gpism, 2021-02-25:
> I update the demo version...you should have another error but maybe you see the blue button
> [image]

Thanks for my internal  demo I need to run it locally on a laptop and ingrate with one data source (like script sending dummy value like oven temperature ) and show various states ...please guide me how do it ...also when it will be fixed ...so that install it NPM package on my system and ingrate with dummy data source for demo

unocelli, 2021-02-25:
you see the blue button?

gpism, 2021-02-25:
Yes I can know see in online demo ...is it fixed in nom version also

unocelli, 2021-02-25:
If you install FUXA by downloading it from the archive [master](https://github.com/frangoteam/FUXA/archive/master.zip) and install it (second option) you have the last and fixed version (not yet released). 
For your demo you can install the the [OCP-UA Server](https://www.unified-automation.com/downloads/opc-ua-servers.html) (freeware) that simulate very good a device connection with changing values.

gpism, 2021-02-26:
> If you install FUXA by downloading it from the archive [master](https://github.com/frangoteam/FUXA/archive/master.zip) and install it (second option) you have the last and fixed version (not yet [...]

Thanks I have  downloaded and used option 2 to install 

So workflow for demo or deployment as per understanding 
1) Create a diagram in Editor 
2) Create  OPC UA source (via stimulator or client libraries)
3) Test with the source and publish 
Is above flow right ?
But how do I embedd or publish FUXA web HMI  to 3rd part site ?

unocelli, 2021-02-27:
Sorry, I don't understand your question?

unocelli, 2021-11-16:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2021-11-16 by unocelli