frangoteam/FUXA #111: Question about a FUXA feature

cyberitot79, 2021-02-25:
Hi there!

I'd like to make a question about one FUXA's functionality I'm having problems with, probable because I'm not launching it as I should be. I've seen FUXA's last versions already have the chance to use Modbus protocol. I'm working on something where I have a simulated PLC using Modbus protocol to communicate, listening in usual port (502) and using normal addresses for the protocol. If I use a normal Modbus client, I can read the values without a problem.

So, I've been trying to use FUXA just to test how it'd be for my environment having an HMI before that PLC device. And honestly FUXA has so many options and features, so it seems a great one to try (you really made a great job!).  I'm a bit new with all of this, especially OT environments, so I wanted to ask something, and I thought this was the best way.

Is it possible to use FUXA as HMI for Modbus devices? I mean, I know the project has the Modbus code for server directory, so I guess it's possible. Once you launch FUXA and you connect 127.0.0.1:1881/device, you can create the server depending on what you want. So I've tried S7 and connection works already. But I was trying to know how to do the same for Modbus. Like, creating a Modbus FUXA Server so I can read my PLC values from there and show them in FUXA environment.

Is it any wiki or similar where Modbus part is explained? When you define Device Property for the server, just S7, OPCUA and WebAPI are the possibilites. And looking at modbus code, I guess to set a Modbus server I should use WebAPI option? I've tried it this way: selecting WebAPI option, so it just let me use GET method and JSON format. Then, I set the URL (I have my modbus device listening on localhost:502), so I guess that's what I need to put in that part. But then... this configuration does not work, it never connects to my device. 

That's why I wonder if there's any place to read more about it, or if you could tell me (if it's not a problem) how I could set a Modbus server, probably I'm doing something wrong.

Really, thank you for your time and your attention, even if you can't help me, I appreciate it!

unocelli, 2021-02-25:
Hi,
To bind Modbus you have to install the plugin if you haven't already. I tested with ModRSsim2
[image]
let me know if you are successful

cyberitot79, 2021-02-25:
Hi!

It was a problem installing the Modbus plugin. I got the project again and reinstalled it, and now I can see Modbus option to create the server. I've tested the connection, and it works when it tries to connect to my port 502 to the slaves. 

But when I try to define a register to read the value, it's not working. FUXA log is showing error every time it tries to read, launching UnhandledPromiseRejectionWarning exception. It says modbus exception 2: Ilegal data address (register not supported by device). I've checked my PLC simulated (I'm using honeypot Conpot for it) and the log shows is sending a bad answer, so I guess that's the reason. When I use the client mbtget against the Conpot it works well, and I've checked the request mbtget sends to the honeypot is different to the one FUXA is sending. I don't know why that could be happening, but I guess it's a problem when the conpot gets the request.

But it worked what you explained about creating the server with Modbus. Thank you so much about that!

unocelli, 2021-02-25:
Thanks for the feedback. 
I'm using the [node-modbus-serial](https://github.com/yaacov/node-modbus-serial) and maybe if we find an example with this driver that works with honeypot Conpot I can understand the problem by FUXA. 
Where I can have honeypot Conpot simulator ?

cyberitot79, 2021-02-25:
Conpot's project is available in here: https://github.com/mushorg/conpot

The honeypot offers so many protocols for PLCs and similar devices. It has a 'default' template that let you execute the honeypot even without needing to add any change or code. That default template puts it to listen on different ports, between them, the 5020 for Modbus (you could need a map with your 502, or just change it in the own template to the port you want). The default template defines slaves with ids 0,255,1 and 2, with different kind of registers, but you can change it to the ones you want just modifying the modbus.xml in the template modbus subdirectory.

I think Conpot is based on modbus-tk for the modbus part, added to their own code. Anything else you need, tell me!

unocelli, 2021-03-04:
Hi,
I installed conpot, I tested with some modbus client to read the register but without success, there is a client that I can see that the master work correctly?

cyberitot79, 2021-03-05:
Hi!

I tested it with this client: https://github.com/sourceperl/mbtget , and if I'm not wrong, it works., at least for registers starting at 30001 or 40001. If you wanna try it in a different way, you can also use this hmi https://github.com/javier-pg/Basic-Modbus-HMI, it reads or writes all the registers without a problem and you just need to tell it the IP and port where you have Conpot.

unocelli, 2021-03-05:
I tested with Basic-Modbus-HMI: I can connect the modbus server and read/write coil from 1 to 128 address, other address make errors, and also make errors every address of digital inputs, inputs registers and holding registers. The same happens with FUXA...

cyberitot79, 2021-03-05:
Hi,

I can read all Conpot registers for Modbus. It will depend on the size you defined for each type of register in the Modbus XML template. Did you check that?

unocelli, 2021-03-14:
Hi,
the default template has already register that I can't read or write, the only ones that I can read are the coils (address Conpot 1 = 2 in FUXA)

<modbus enabled="True" host="0.0.0.0" port="5020">
    <device_info>
        <VendorName>Siemens</VendorName>
        <ProductCode>SIMATIC</ProductCode>
        <MajorMinorRevision>S7-200</MajorMinorRevision>
    </device_info>
    <mode>serial</mode>
    <delay>100</delay>
    <slaves>
        <slave id="0">
            <blocks>
                <block name="memoryModbusSlave0BlockA">
                    
                    <type>COILS</type>
                    <starting_address>1</starting_address>
                    <size>128</size>
                    <content>memoryModbusSlave0BlockA</content>
                </block>
                <block name="memoryModbusSlave0BlockB">
                    
                    <type>DISCRETE_INPUTS</type>
                    <starting_address>10001</starting_address>
                    <size>32</size>
                    <content>memoryModbusSlave0BlockB</content>
                </block>
            </blocks>
        </slave>
        <slave id="255">
            <blocks>
                <block name="memoryModbusSlave255BlockA">
                    
                    <type>COILS</type>
                    <starting_address>1</starting_address>
                    <size>128</size>
                    <content>memoryModbusSlave255BlockA</content>
                </block>
                <block name="memoryModbusSlave255BlockB">
                    
                    <type>DISCRETE_INPUTS</type>
                    <starting_address>10001</starting_address>
                    <size>32</size>
                    <content>memoryModbusSlave255BlockB</content>
                </block>
            </blocks>
        </slave>
        <slave id="1">
            <blocks>
                <block name="memoryModbusSlave1BlockA">
                    
                    <type>COILS</type>
                    <starting_address>1</starting_address>
                    <size>128</size>
                    <content>memoryModbusSlave1BlockA</content>
                </block>
                <block name="memoryModbusSlave1BlockB">
                    
                    <type>DISCRETE_INPUTS</type>
                    <starting_address>10001</starting_address>
                    <size>32</size>
                    <content>memoryModbusSlave1BlockB</content>
                </block>
            </blocks>
        </slave>
        <slave id="2">
            
            <blocks>
                <block name="memoryModbusSlave2BlockC">
                    
                    <type>ANALOG_INPUTS</type>
                    <starting_address>30001</starting_address>
                    <size>8</size>
                    <content>memoryModbusSlave2BlockC</content>
                </block>
                <block name="memoryModbusSlave2BlockD">
                    
                    <type>HOLDING_REGISTERS</type>
                    <starting_address>40001</starting_address>
                    <size>8</size>
                    <content>memoryModbusSlave2BlockD</content>
                </block>
            </blocks>
        </slave>
    </slaves>
</modbus>

cyberitot79, 2021-03-18:
And what is the error Conpot throws you when you try to read them?

unocelli, 2021-03-21:
I don't know what i make wrong, I installed again and started:

- docker pull honeynet/conpot
- docker run -it -p 80:80 -p 102:102 -p 502:502 -p 161:161/udp --network=bridge honeynet/conpot:latest /bin/sh

docker logs:
[image]

I tried with 2 client, I can connect it but now I can't even read the coils anymore :(
[image]

[image]

unocelli, 2021-05-07:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2021-05-07 by unocelli