frangoteam/FUXA #113: Chart feature

zesava, 2021-03-03:
Is it possible to have a few Y axes ?
This may be useful when need to compare data with different scale, eg when tuning PID regulator. Where PV and SP may change from 0 to 10, and MV changes from 0 to 100%

Let's say something similar to this
https://stackblitz.com/edit/angular-gqnztk?file=src/app/app.component.ts

unocelli, 2021-03-04:
Hi,
the dygraphs used in FUXA support [2 Y-axes](https://dygraphs.com/tests/two-axes.html). We can add in chart settings and eventually add the synchronization between more charts. 
I think is not so easy to find a charts library rich of feature, performant that pleases everyone. [uPlot](https://github.com/leeoniya/uPlot) is a candidate which I would like to integrate into FUXA.

unocelli, 2021-05-01:
I changed to uPlot and added the Y axes feature [uPlot chart with y-axis and history](https://github.com/frangoteam/FUXA/commit/716a17f94c9a029a6fd1d8e96236f5e01d994c88)

closed 2021-05-07 by unocelli