frangoteam/FUXA #114: Browse Tags in BACnet device causes a crash

madmax76, 2021-03-04:
Platform: OSX 11.2.2
Node: v15.10.0
FUXA: v1.0.9

```terminal
2021-03-04T07:54:31.576Z [ERR] 	'bacnet2' _askName error! Error: ERR_TIMEOUT
    at Timeout._onTimeout (/usr/local/lib/node_modules/@frangoteam/fuxa/_pkg/bacstack/lib/client.js:76:16)
    at listOnTimeout (node:internal/timers:557:17)
    at processTimers (node:internal/timers:500:7)
2021-03-04T07:54:31.577Z [ERR] 	UnhandledPromiseRejection: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason "undefined".
```

I have made some pcaps, but I am unable to share them on a public forum. Someone willing and able to help on this?

unocelli, 2021-03-05:
Hi,
can you tell me if you have the same error also with node v14 or 12?

matrix926, 2021-03-07:
> 
> 
> Hi,
> can you tell me if you have the same error also with node v14 or 12?
node v12.21.0

2021-03-07T14:49:59.618Z [ERR]      'Bacnet' working (connection || polling) overload!

madmax76, 2021-03-09:
@unocelli :

**Version: v12.21.0**
```console
POST /api/projectData 200 27.081 ms - -
2021-03-09T21:10:45.673Z [ERR] 	'virtualbox' _askName error! Error: ERR_TIMEOUT
    at Timeout._onTimeout (/usr/local/lib/node_modules/@frangoteam/fuxa/_pkg/bacstack/lib/client.js:76:16)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
(node:65659) UnhandledPromiseRejectionWarning: undefined
(node:65659) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:65659) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

**Version: v14.16.0**
With this version getting a bit further, it lists the objects, but while expanding them, the following message appears:

```console
GET /api/plugins 304 9.167 ms - -
2021-03-09T21:19:34.359Z [ERR] 	'virtualbox' working (connection || polling) overload!
2021-03-09T21:19:36.361Z [ERR] 	'virtualbox' working (connection || polling) overload!
2021-03-09T21:19:38.365Z [ERR] 	'virtualbox' working (connection || polling) overload!
2021-03-09T21:19:40.368Z [ERR] 	'virtualbox' working (connection || polling) overload!
2021-03-09T21:19:42.369Z [ERR] 	'virtualbox' working (connection || polling) overload!
2021-03-09T21:19:44.374Z [ERR] 	'virtualbox' working (connection || polling) overload!
2021-03-09T21:19:46.379Z [ERR] 	'virtualbox' working (connection || polling) overload!
```
It is not crashing, but the overload message keeps repeating itself. Will a pcap help with the investigation?

[image]

unocelli, 2021-03-14:
Hi,
I think I fixed it.
@madmax76: better options as plugin... you have an idea?

madmax76, 2021-03-15:
@unocelli : **The good news:** It seems no longer to terminally crash, good work. **The bad news**, from what I see:

1. application does a readProperty -> object-name (77)
2. BACnet device answers -> reject - unrecognized - service (67)
3. application does not catch the exception of the BACnet not handling the service.
4. application jumps out of the loop reading the objects.

```terminal
2021-03-15T07:38:21.986Z [ERR] 	'test' _askName error! Error: BacnetError - Class:5 - Code:67
    at processError (/Users/*****/code/beta-fuxa/FUXA/server/_pkg/bacstack/lib/client.js:96:30)
    at handlePdu (/Users/*****/code/beta-fuxa/FUXA/server/_pkg/bacstack/lib/client.js:309:9)
    at handleNpdu (/Users/*****/code/beta-fuxa/FUXA/server/_pkg/bacstack/lib/client.js:343:5)
    at self.receiveData (/Users/*****/code/beta-fuxa/FUXA/server/_pkg/bacstack/lib/client.js:354:7)
    at Socket.<anonymous> (/Users/*****/code/beta-fuxa/FUXA/server/_pkg/bacstack/lib/transport.js:14:25)
    at Socket.emit (events.js:314:20)
    at UDP.onMessage [as onmessage] (dgram.js:921:8)
(node:34622) UnhandledPromiseRejectionWarning: undefined
(node:34622) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
```

tested with node versions 10, 12 & 14
alternative bacnet stack: [Node-bacstack by BiancoRoyal](https://github.com/BiancoRoyal/node-bacstack
)

unocelli, 2021-03-15:
Hi,
my resources are limited to test it because I don't use BACnet in my devices. I tested with [ProtoSense BACnet device simulator](https://protosensetech.com/downloads/bacnetip_device_simulator) where it works and I can't reproduce the error. You know some freeware simulator to better test and reproduce the error ?
I'll try the BiancoRoyal, it looking more popular...Thanks

madmax76, 2021-03-21:
> Hi,
> my resources are limited to test it because I don't use BACnet in my devices. I tested with [ProtoSense BACnet device simulator](https://protosensetech.com/downloads/bacnetip_device_simulator) [...]

I am using [YABE - Yet Another Bacnet Explorer](https://sourceforge.net/projects/yetanotherbacnetexplorer/) It needs windows, but after installing there is a room-thermostat simulator and two other simulators in c:\Program Files\YABE\Addons.

Thank you for the ProtoSense hint.

unocelli, 2021-03-23:
@madmax76 Thanks, with the YABE simulator I was able to reproduce the error.
I fixed, the error on browse (askName), I would be grateful if I could confirm me ;)

madmax76, 2021-03-23:
@unocelli Yes, it works now, thank you. Great work, it works really well now and the project is coming alive to me. Continuing with creating some HMI's and then continue with S7 & OPC UA testing.

Sirezul, 2021-04-27:
Hi, first of all awesome Project! :)

After adding my bacnet device, I'm also facing crashes while browsing it:

`2021-04-27T14:56:46.690Z [ERR]  'test' _askName error! Error: ERR_TIMEOUT
    at Timeout._onTimeout (D:\Projekte\Fuxa\FUXA-1.0.10\server\_pkg\bacstack\lib\client.js:76:16)
    at listOnTimeout (internal/timers.js:551:17)
    at processTimers (internal/timers.js:494:7)
(node:2796) UnhandledPromiseRejectionWarning: undefined
(Use `node --trace-warnings ...` to show where the warning was created)
(node:2796) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:2796) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

unocelli, 2021-04-30:
with which version of FUXA are you trying? the last one of master branch? [6b54e46](https://github.com/frangoteam/FUXA/commit/6b54e464d3b42e46ddd80c799b9eda5ce4c5aeee)

Sirezul, 2021-05-04:
Hi, sorry for the late replay. I've used the latest [release ](https://github.com/frangoteam/FUXA/releases/tag/v1.0.10) version.
With the fix from this issue.

Sirezul, 2021-05-04:
Okay It was my fault! 😐 After cloning it new, it works! Thank you.

unocelli, 2021-05-04:
@Sirezul Thank you for the feedback.
I'm going to close this as resolved. let me know if you have any issues.

closed 2021-05-04 by unocelli