frangoteam/FUXA #116: Error

danlesbm, 2021-03-29:
Hi frangoteam
I am trying FUXA. But it had some errors as follows
1. Graph: The value isn't stable. But value show at sceen design, it ok. I think it has undefined values that you don't remove from the graph.
2. Alarm: It doesn''t work with API connection while TCP is ok
3. Device: Frequent connection failures if set timeout is short. Specifically: if left less than 30 seconds, there will be errors
4. The password is set, but the default start-up page is still visible to everyone
5. if you can add the tabular history and can export  * .csv file. I think it will be better 
I feel the software very good, if it fixes the above errors 
Thank you and Best regards!
[image]

[image]
[image]

danlesbm, 2021-03-31:
I reviewed Graph carefully, i think the reasons:
if using many parameters in 1 graph: 
- This parameter mistakenly takes the value of another parameter 
-  At different times, have undefined parameters
 -> show error

[image]

unocelli, 2021-04-04:
Hi, thanks for your report and I'm looking in it...I would be thankful if you can split in 5 issues ;)

danlesbm, 2021-04-04:
> Hi, thanks for your report and I'm looking in it...I would be thankful if you can split in 5 issues ;)

Ok. I have splitted

unocelli, 2021-05-02:
Graph: The value isn't stable #124
Alarm: It doesn''t work with API connection while TCP is ok #125
Device: Frequent connection failures #126
Set password #127
export * .csv file #128

closed 2021-05-02 by unocelli