frangoteam/FUXA #117: requirements of data upload

zjjczzs, 2021-03-29:
I want to transfer data to cloud platform through mqtt, such as thingsboard. Can this function be realized?

unocelli, 2021-04-01:
What is your goal?
FUXA make process visualization, I think there are better software than FUXA to act as a gateway.

zjjczzs, 2021-04-03:
In some small data acquisition cases, gateways, HMI, and data processing are expected to be integrated into a device for cost reasons. I've had cases like this in an actual project. FUXA, on the other part, has the potential to do just that. Thank you!

zjjczzs, 2021-04-03:
> What is your goal?
> FUXA make process visualization, I think there are better software than FUXA to act as a gateway.

In some small data acquisition cases, gateways, HMI, and data processing are expected to be integrated into a device for cost reasons. I've had cases like this in an actual project. FUXA, on the other part, has the potential to do just that. Thank you!

unocelli, 2021-08-07:
Hi,
In case you're still interested, I've made some changes in MQTTclient, so you can now publish tags from other devices.

unocelli, 2021-11-16:
This feature was implemented. I will therefore close this issue, feel free to re-open it if something is still not working.

closed 2021-11-16 by unocelli

zjjczzs, 2021-11-24:
I tested using mqtt to publish and subscribe. The subscription function works well, but the publishing function doesn't work. In addition, I updated zh-cn.json

从 Windows 版邮件发送

发件人: Umberto Nocelli
发送时间: 2021年11月17日 5:37
收件人: frangoteam/FUXA
抄送: zjjczzs; Author
主题: Re: [frangoteam/FUXA] requirements of data upload (#117)

This feature was implemented. I will therefore close this issue, feel free to re-open it if something is still not working.

unocelli, 2021-11-24:
Hi,
I'm coming from test the publish and i don't see problem. can you send me how you have configured it?
about language update you can make a PR or send me the file

zjjczzs, 2021-11-25:
I created an mqtt server, which can be used in https://cloud-intl.emqx.com/ Login, user name: ***@***.*** , password: o000000o

从 Windows 版邮件发送

发件人: Umberto Nocelli
发送时间: 2021年11月25日 3:20
收件人: frangoteam/FUXA
抄送: zjjczzs; Author
主题: Re: [frangoteam/FUXA] requirements of data upload (#117)

Hi,
I'm coming from test the publish and i don't see problem. can you send me how you have configured it?
about language update you can make a PR or send me the file

zjjczzs, 2021-11-25:
I accidentally replaced the demo in GitHub. Please check and restore it

从 Windows 版邮件发送

发件人: Umberto Nocelli
发送时间: 2021年11月25日 3:20
收件人: frangoteam/FUXA
抄送: zjjczzs; Author
主题: Re: [frangoteam/FUXA] requirements of data upload (#117)

Hi,
I'm coming from test the publish and i don't see problem. can you send me how you have configured it?
about language update you can make a PR or send me the file

unocelli, 2021-11-25:
Don't worry the demo save the project only in your browser (localStorage)

I try to publish a opcua tag value and it work really good...
[image]

[image]

zjjczzs, 2021-11-26:
I public the Modbus connection data of tag type and it can work. However, using the three types of constant, topic and timestamp in the publishing type, there is no response or prompt after clicking the Publish button.
    In addition, during the test, it was found that once I disconnected the connection in the modsim, the corresponding connection in fuxa turned red and an error prompt popped up. After restoring the modsim connection, fuxa could not automatically reconnect, and the red and pop-up error prompt were always displayed.

从 Windows 版邮件发送

发件人: Umberto Nocelli
发送时间: 2021年11月26日 4:29
收件人: frangoteam/FUXA
抄送: zjjczzs; Author
主题: Re: [frangoteam/FUXA] requirements of data upload (#117)

Don't worry the demo save the project only in your browser (localStorage)
I try to publish a opcua tag value and it work really good...

unocelli, 2021-11-26:
Can you send me a screenshot of the data that you can not publish please?

zjjczzs, 2021-11-27:
从 Windows 版邮件发送

发件人: Umberto Nocelli
发送时间: 2021年11月27日 2:24
收件人: frangoteam/FUXA
抄送: zjjczzs; Author
主题: Re: [frangoteam/FUXA] requirements of data upload (#117)

Can you send me a screenshot of the data that you can not publish please?