frangoteam/FUXA #118: Allen Bradley

Scolari, 2021-03-29:
How can I connect a Compactologix, Allen Bradley PLC?

unocelli, 2021-04-01:
Hi,
Allen Bradley PLC are not supported yet, I think with [node-ethernet-ip](https://github.com/cmseaton42/node-ethernet-ip) can work but as already discussed in #72 in order to implement this connection we have to be able to test with a PLC or a simulator that we don't have.

fedex-12, 2021-04-14:
Hello,
node-ethernet-ip is not updated.
This project is updated: [https://github.com/libplctag/libplctag](https://github.com/libplctag/libplctag)
If you can develop it I can test it with the PLC.

unocelli, 2021-04-16:
Hi,
the library libplctag  is in C, we need in javascript to integrate it into a nodejs project :(

concept10, 2021-05-06:
@unocelli Was wondering if you ever considered wrapping C based libraries in node using N-API (or simular functionality) or possibly compiling some C/C++ based communications library using WebAssembly?  Just curious.

unocelli, 2021-05-09:
@concept10 very interesting and I would like to go further, probably later. Currently I would like to invest resources in the frontend.

concept10, 2021-05-14:
@unocelli Sure thing.  I've been in this industry for 30 years this year.  What you've done so far is excellent (on both the backend and frontend).  This is from someone who has worked with all major manufacturers.  When I get additional time, I will join you to help as needed and looking forward to it.

unocelli, 2021-05-15:
@concept10 Thanks for the appreciation, I need help, you are welcome.

unocelli, 2021-11-21:
Is there anyone who would like to help me by test the support of Allen Bradley plc with the 'ethernetip' branche.

AlexCastro-Prog, 2021-11-22:
Hello Umberto,

I can help you.

I was helping you with nodepccc and I can help with Ethernet/IP. I have all PLCs and I was waiting your invite to get access and get progress.

I'm waiting from you.

Regards,

Alex

Sent from my Verizon, Samsung Galaxy smartphone

concept10, 2021-11-23:
@AlexCastro-Prog Have you ever tested against the Allen Bradley simulator (made by OEM).

concept10, 2021-11-23:
I asked about the testing and willing to help as well. I'm just thinking about the plan for testing. 

I've been promoting FUXA to customers along with the OEM solutions.  Trying to get you guys some shine in manufacturing (automotive, semiconductor) and I sometimes have to show them how the software was tested.

AlexCastro-Prog, 2021-11-24:
Hello team,

With Rockwell controllers is not possible to do a test with simulator because the simulator does not have an IP Address.

The test needs to be with real PLC so in this way we know that will work as decided.

Regards,

Alex Castro

AlexCastro-Prog, 2021-11-30:
[index.zip](https://github.com/frangoteam/FUXA/files/7622367/index.zip)
Hello Umberto here is the FUXA/server/runtime/devices/ethernetip/index.js that we worked last Sunday.

I appreciate all your effort to make this work.

Thank you so much!

cjmccoy2, 2021-12-21:
@AlexCastro-Prog Do you have any documentation on using the driver? I am having issues getting it to work

AlexCastro-Prog, 2021-12-27:
> @AlexCastro-Prog Do you have any documentation on using the driver? I am having issues getting it to work

Hello @cjmccoy2 I'll post a video this week related to this it. What is the problem that you have been seen with Ethernet/IP node?

cjmccoy2, 2021-12-27:
Hello @AlexCastro-Prog . I am unable to get any data from B3:01/0 file types. I also cannot get any tags from a micro800 to read. This is the most important. Thanks!

unocelli, 2022-02-15:
going to close this as resolved. let me know if you have any issues.

closed 2022-02-15 by unocelli

jpomo, 2023-06-22:
@unocelli @AlexCastro-Prog Thanks for the explanation video.

Concerning the routing settings, I use an AB CompactLogix PLC which has the ethernet ports built into the PLC (not in a slot on a backplane). Do you have any thoughts on how I can set the routing?

[image]