frangoteam/FUXA #121: Can you provide relevant documents？

feiyiban, 2021-04-01:
can you add some documents to explain？

unocelli, 2021-04-01:
you mean a manual of how to use it? have you already looked on the wiki?

feiyiban, 2021-04-02:
I'm very sorry. I just saw your project. I feel great. I want to have a complete understanding of the whole project. I haven't seen wiki yet. Recently, I encountered a problem. I use Siemens S7 200, but it seems that I can't connect. I don't know if your protocol supports 200 devices. thank you

feiyiban, 2021-04-02:
I'm very sorry. I looked at the wiki. I mean, can the document be richer, the table design, the overall framework, and whether it can support Chinese

unocelli, 2021-04-02:
You are right the wiki could be richer, I admit that I prefer to focus on the features ;) I'm waiting for some help
For the support of Chinese you should make a file as for Russian and then a pull request like #107 or #112 ;), if you look in 'client/src/assets/i18' you should understand.

feiyiban, 2021-04-06:
I'm familiar with nodejs now. Maybe I don't know much about your project at present. I'll refuel. Thank you

closed 2021-04-06 by feiyiban