frangoteam/FUXA #124: Graph: The value isn't stable

danlesbm, 2021-04-04:
The value isn't stable. But value show at sceen design, it ok. I think it has undefined values that you don't remove from the graph
I reviewed Graph carefully, i think the reasons:
if using many parameters in 1 graph:

This parameter mistakenly takes the value of another parameter
At different times, have undefined parameters
-> show error
in the photo, i show 2 parameters. The red color, value ranges: 110-116, but some time at here value = 0, some time it''s true = 100-116
The blue color, value ranges: 49-51, but some time at here value = 0, some time > 400, some time it''s true = 49-51
[image]

unocelli, 2021-04-19:
Hi,
sorry for the delay. 
You mean that the error is in the chart representation or saved wrong in the database? Which device are you using?
To compare, can you add two charts with only one of the line?

unocelli, 2021-05-07:
I changed the chart control to uPlot. I assume the issue has been resolved.

danlesbm, 2021-05-18:
> I changed the chart control to uPlot. I assume the issue has been resolved.

I see this error is gone. thanks

closed 2021-05-18 by danlesbm