frangoteam/FUXA #126: Device: Frequent connection failures

danlesbm, 2021-04-04:
Frequent connection failures if set timeout is short. Specifically: if left less than 30 seconds, there will be errors
[image]

unocelli, 2021-04-07:
I assume this screenshot is not relevant...with which type of connection? can you send me the exception message?

danlesbm, 2021-04-08:
> I assume this screenshot is not relevant...with which type of connection? can you send me the exception message?

OK. I uploaded the wrong picture. below is the video, if left less than 30 seconds or after reboot, there will be errors

https://user-images.githubusercontent.com/65875578/113951901-be1e8e80-983e-11eb-9911-6a7add254214.mp4

unocelli, 2021-04-08:
Thanks, i found a concept error, after a request error don't try again, I fixed and i added the missing error log.
can you try and send me log file '_log/fuxa-err.log'

danlesbm, 2021-04-08:
> Thanks, i found a concept error, after a request error don't try again, I fixed and i added the missing error log.
> can you try and send me log file '_log/fuxa-err.log'

ok

danlesbm, 2021-04-09:
It still didn't run. It running, i reboot my PC, after reconnect but didn't run. I must push modify then push ok --> it run.
I didn't find file log

unocelli, 2021-04-09:
I'm not sure if you did the update. 
It's better if you test the WebAPI alone,  install FUXA again from [master branch](https://github.com/frangoteam/FUXA/archive/refs/heads/master.zip) or 'git clone https://github.com/frangoteam/FUXA'
start the server, configure only the WebAPI and then you have to find the error log in the folder where you have started the server 'FUXA/server/_logs'.

danlesbm, 2021-04-09:
> I'm not sure if you did the update.
> It's better if you test the WebAPI alone, install FUXA again from [master branch](https://github.com/frangoteam/FUXA/archive/refs/heads/master.zip) or 'git clone [...]

yes of course, i ran it with command: "npm install -g --unsafe-perm @frangoteam/fuxa"

unocelli, 2021-04-09:
By npm server you install only the released version and I have not yet released with the latest changes.
To be update you have to install from github

danlesbm, 2021-04-09:
> By npm server you install only the released version and I have not yet released with the latest changes.
> To be update you have to install from github

OK. Let me try again

unocelli, 2021-04-09:
You have to: 
- download from [master branch](https://github.com/frangoteam/FUXA/archive/refs/heads/master.zip)
- cd ./server
- npm install
- npm start

danlesbm, 2021-04-09:
> You have to:
> 
> * download from [master branch](https://github.com/frangoteam/FUXA/archive/refs/heads/master.zip)
> * cd ./server
> * npm install
> * npm start

I tested, it's ok

unocelli, 2021-04-09:
great, you can close the solved issues

closed 2021-04-09 by danlesbm