frangoteam/FUXA #130: about Internationalization

zjjczzs, 2021-04-09:
I compiled a document for Chinese translation, named ch.json , and copy it to the folder of ⁃ client ⁃ SRC ⁃ assets ⁃ I18N, and then what can I do to load the new language in the settings?

unocelli, 2021-04-09:
You have to add in 'client/src/app/editor/app-settings/app-settings.component.ts' to 'languageType' array variable:
` { text: 'dlg.app-language-ch', value: 'ch' }`

zjjczzs, 2021-04-17:
> You have to add in 'client/src/app/editor/app-settings/app-settings.component.ts' to 'languageType' array variable:
> ` { text: 'dlg.app-language-ch', value: 'ch' }`

I've added array elements, but that doesn't work out：
 languageType = [ { text: 'dlg.app-language-en', value: 'en' }, { text: 'dlg.app-language-ru', value: 'ru' }, { text: 'dlg.app-language-ua', value: 'ua' }, { text: 'dlg.app-language-ch', value: 'ch' } ];

unocelli, 2021-04-17:
You have to compile the client:
`cd .\client`
`npm install`
`ng build --prod`

zjjczzs, 2021-04-19:
> You have to compile the client:
> `cd .\client`
> `npm install`
> `ng build --prod`

I've compiled the client and I can choose Chinese in the settings menu, but the interface is still in English, what should I do next?

unocelli, 2021-04-19:
You can send me the ch.json file so that I can try

zjjczzs, 2021-04-22:
> You can send me the ch.json file so that I can try

Has been sent to the FUXA@noreply.github.com, please check

unocelli, 2021-04-22:
Sorry, I don't have a FUXA@noreaply.github.com email, please send me to 4frango@gmail.com

zjjczzs, 2021-04-23:
Translation is immature, need to be further improved according to hmi, so please provide me with a test environment, thank you!

Sorry, I don't have a ***@***.*** email, please send me to ***@***.***

unocelli, 2021-04-23:
You have in your 'zh.cn.json' file at line 448 a syntax error  """ " you have to change with "": ""
don't forget to add ` { text: 'dlg.app-language-zh-ch', value: 'zh-cn' }` at `languageType` variable in 'app-settings.component.ts'
It's work for me :)

unocelli, 2021-04-25:
going to close this as resolved. let me know if you have any issues.

closed 2021-04-25 by unocelli

zjjczzs, 2021-04-26:
ok

Closed #130.

unocelli, 2021-04-28:
Good work, Thanks a lot :D

zjjczzs, 2021-09-10:
Add：, { text: 'dlg.app-language-zh-cn', value: 'zh-cn'}
Translation is immature and needs to be further refined according to hmi

发送自 Windows 10 版邮件应用

发件人: Umberto Nocelli
发送时间: 2021年4月20日 0:06
收件人: frangoteam/FUXA
抄送: zjjczzs; Author
主题: Re: [frangoteam/FUXA] about Internationalization (#130)

You can send me the ch.json file so that I can try