frangoteam/FUXA #131: MQTT doesn't work

fedex-12, 2021-04-14:
Hello,
i used the MQTT driver, but it doesn't work.
I tried to connect to the broker: broker.hivemq.com
Is there a guide?
Thank you.

unocelli, 2021-04-19:
Hi,
sorry for the delay, which error you receive?

amaloma, 2021-04-24:
Me too
[image]
[image]
[image]

unocelli, 2021-04-24:
Thanks for the report, I'm looking for

unocelli, 2021-04-24:
Fixed

amaloma, 2021-04-25:
connect ok, but it doesn't work to subscribe.

[image]
[image]
[image]
[image]
[image]

unocelli, 2021-04-25:
@amaloma Thanks for the report, I'm looking for

unocelli, 2021-04-25:
Fixed

amaloma, 2021-04-26:
It works, Thanks.

unocelli, 2021-04-30:
going to close this as resolved. let me know if you have any issues.

closed 2021-04-30 by unocelli