frangoteam/FUXA #134: MQTT json data

amaloma, 2021-04-25:
Please add mqtt json key to tag name

e.g. 
1. topic = my/test/topic, message = 55
2. topic = test/topic/sensor, message = { "Temperature":  25, "Humidity": 40 }

[image]

unocelli, 2021-04-26:
I understand your request...what should I send to the broker if in FUXA with a button I change only the 'Temperature' value?
if I send only the json with 'Temperature' I lost the other attribute ('Humidity') in 'test/topic'?

unocelli, 2021-08-07:
Hi,
I make some improvments in MQTT client for the publish functionality...

MRIIOT, 2021-08-20:
In my case my payload looks like this:

```
[{"address": "%M55.0", "keys": ["counter.0.value"], "value": 778, "_ts": 1629498395587}]
```

unocelli, 2021-08-22:
Hi,
where the expression check should be done? in the backend? is something you want to be able to configure to every topics?

MRIIOT, 2021-08-23:
> Hi,
> where the expression check should be done? in the backend? is something you want to be able to configure to every topics?

Ideally it would be done when binding control.  In my case, different I/O data arrives on the same topic. One control might be bound to value of %M55.0, another to value of %M55.1.

I don't know that JSONata is the correct answer as it only works with JSON.  Are there ways to extend Node with another scripting language (eg JS, Lua, Python) to give user ability to write scripts?  Maybe 'safe-eval' https://github.com/hacksparrow/safe-eval.

unocelli, 2021-09-18:
Hi,
I did some refactoring in MQTT client...maybe solve the issues.

unocelli, 2022-01-30:
going to close this as resolved. let me know if you have any issues.

closed 2022-01-30 by unocelli