frangoteam/FUXA #136: Navigation menu

airtw, 2021-05-01:
When the navigation menu is an external link, logging out cannot effectively clear the screen content
https://upload.cc/i1/2021/05/01/94hWLp.png

unocelli, 2021-05-01:
which error is shown in browser console?

airtw, 2021-05-01:
No error, no relevant measures for logout operations.
I tried to solve this way.
```
	onLogin() {
		let cuser = this.authService.getUser();
		if (cuser) {
			let dialogRef = this.dialog.open(DialogUserInfo, {
				id: 'myuserinfo',
				minWidth: '250px',
				position: { top: '50px', right: '15px' },
				backdropClass: 'user-info',
				data: cuser
			});
			dialogRef.afterClosed().subscribe(result => {
				if (result) {
					if(this.iframeview != null){
						this.showHomeView = true;
						this.showHomeLink = false;
						this.changeDetector.detectChanges();
					}
					this.authService.signOut();
					this.projectService.reload();
				}
			});
		} else {
			let dialogRef = this.dialog.open(LoginComponent, {
				minWidth: '250px',
				data: {}
			});
			dialogRef.afterClosed().subscribe(result => {
			});
		}
	}
```

unocelli, 2021-05-02:
I don't understand what is your problem or goal?

unocelli, 2021-11-16:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2021-11-16 by unocelli