frangoteam/FUXA #137: Modbus data decimals

T1bbY, 2021-05-03:
Hi,

I work on an interface for my heatpump and i get the data through modbus tcp.
That works fine :)

But the values are send as integer like here the temperature: 509

[image]

But the temperature is 50,9.
Under "Interativity"->"Property" ->"Format digits", FUXA only adds ".0" to my value.

How can i tell FUXA to do this?

Thanks for you help :)

unocelli, 2021-05-04:
Hi,
You are right in modbus is missing, I added the feature, You can now set a divisor in tag property of device, so we have the converted value for all controls. [Modbus Tags with divisor](https://github.com/frangoteam/FUXA/commit/11df175afc4b30bb1c82152b1e1a48827effa9aa) 
[image]

T1bbY, 2021-05-04:
Wow, that was fast :)

Problem solved :)
[image]

closed 2021-05-04 by T1bbY