frangoteam/FUXA #138: Question: BACnet Stack

Sirezul, 2021-05-05:
Hi, just out of curiosity. Why is the version of the bacnet-stack from [fh1ch](https://github.com/fh1ch/node-bacstack) used?

It seems like [this](https://github.com/BiancoRoyal/node-bacstack#readme) is the successor of the project. Even though right now I don't know if there is an advantage. 😊

unocelli, 2021-05-05:
Hi,
I have to be honest, I looked at the popularity 'Star' but maybe that was not a good decision.
I don't use BACnet and I have only a dummy tool as simulator. We can try with the [node-bacstack](https://github.com/BiancoRoyal/node-bacstack), the api shouldn't be very different. what do you think about? would you be willing to test it?

Sirezul, 2021-05-05:
As I mainly want to use fuxa for BACnet devices I'll probably give it a try. I'll keep you updated :)

unocelli, 2021-05-14:
Sorry for the delay. 
I will try to look the difference by change the library and as soon as I have something I will keep you informed.

unocelli, 2021-05-24:
@Sirezul Hi, I added the node-bacnet library, let me know if you have any issues.

Sirezul, 2021-05-25:
Hi @unocelli, I'll test asap and provide feedback. Thank you very much! 👍

vangola-silver, 2021-06-01:
Hi @unocelli, thanks for your amazing app. 
I have tried using FUXA 1.0.11 to connect to a Bacnet device however, it stucks at "Browse Tags in Server". 

**Used Software:**
Node: 14.17.0
npm: 7.15.1
Yabe (Room Controll Simulator)
Tridium Niagara: 4.9.0.198

[image]

The console shows the following:

`2021-06-01T07:10:52.554Z [WAR]  'test' working (connection || polling) overload!
2021-06-01T07:10:53.555Z [WAR]  'test' working (connection || polling) overload!
2021-06-01T07:10:54.555Z [WAR]  'test' working (connection || polling) overload!
2021-06-01T07:10:55.170Z [ERR]  'test' _readObjectList error! Error: ERR_TIMEOUT
2021-06-01T07:10:55.555Z [WAR]  'test' working (connection || polling) overload!
2021-06-01T07:10:56.556Z [WAR]  'test' working (connection || polling) overload!
`

I have also tried using the new BACnet stack, but I could not go any forther. 
I believe the issue is somehow connected to this issue #114. 

@Sirezul Did you manage to test it yet?

unocelli, 2021-06-01:
Hi,
please can you try with the last master version of the repository (V1.0.12-beta)?
I changed BACnet library to node-bacnet and maybe it works better.

vangola-silver, 2021-06-07:
Hi @unocelli,

I have tried to use the latest master version, but I am facing some difficulties trying to install the npm dependencies. 
As you can see on the screenshot npm complains about deprecated dependencies. However, I could not figure out which module requires all those deprecated modules. Do you perhaps have any ideas on this?

[image]

unocelli, 2021-06-07:
Hi,
Please can you send me the os and npm version?

vangola-silver, 2021-06-07:
NPM: 7.15.1
OS: Windows 10 1909 Build 18363.1379

unocelli, 2021-06-07:
I don't understand why the installation uses python :(
Have you in windows a microsoft visual c++ redistributable version installed?

vangola-silver, 2021-06-07:
Yes I have.

unocelli, 2021-06-07:
yes, now I remember having already had such an issue.
the node-pre-gyp used from the node-snap7 library is not compatible with the npm 7.x version.
We should open a issue about this.
Thanks

vangola-silver, 2021-06-08:
As a workaround I did some juggle with the environment variables so it can find my python 2.7 installation. Googling “node-gyp window python error” I found out that the old node-gyp version required python 2.7.

unocelli, 2021-06-08:
Thanks.
I'm confused...I have just installed fuxa on a new windows 10 after installing node version 14.17.0 with includes npm 6.14.13, python are not installed and not required :/

hongnhan100, 2021-08-14:
Hello @unocelli.
I have used both node-Bacnet and bacstack with Delta Control device.
Node-bacnet can't get objectList from device.
[image]
[image]

hongnhan100, 2021-08-14:
I changed file _server\runtime\devices\bacnet\index,js_ to use **"bacstack": "0.0.1-beta.14"**.  It works better.
And I get units (AV,AO,AI), Active_Text\Inactive_Text(BV,BI,BO) from Object(**readPropertyMultiple function**).
Can You help me add them to UI?
[image]

unocelli, 2021-08-16:
@hongnhan100 Hi, backstack was my first choice, changing to node-bacnet was a test that evidently failed. I saw that the repository is active again. I will change back to bacstack (0.0.1-beta.14). Let me know if the issue to add units column is still current. Thanks

hongnhan100, 2021-08-16:
Hi @unocelli 
I have successfully added the unit to the tag. And change Value to On/Off for BV,BO,BI tag. 
I will change them to Active/Inactive Text read from bacnet point.
[image]

unocelli, 2021-08-16:
@hongnhan100 good work, then I do nothing and I wait your PR ;)

hongnhan100, 2021-08-16:
In the future, I want Units from Output Setting can get from Tag.
And Value can display Inactive/Active Text.
[image]

hongnhan100, 2021-08-16:
Can I send you file index.js (Using bacstack 0.0.1-beta.14)?

unocelli, 2021-08-16:
The problem is that than you are not flexible for other type of device where the unit came from other tags

unocelli, 2021-08-16:
Yes send me, thanks

hongnhan100, 2021-08-16:
> Yes send me, thanks

[bacnet.zip](https://github.com/frangoteam/FUXA/files/6994396/bacnet.zip)

Xinmer, 2022-01-07:
Have you ever tried to add multiple Bacnet devices? Do you support this?
I am now testing to add two BACnet objects, but only one can pass data, and the other will be blocked。
Can you help me?

katzenknilch, 2022-01-23:
Hey together,

I tried also to connect multiple devices in a Network.
If I write the IP of my Computer in the Node-bacnet client.js at LOCL_HOST_IPv4 and add a Bacnet Device with the localhost IP, so it show all Devices in the Network.

But, my Problem is that I  got polling overload.
So I want to know how to increase the Pollingrate over 5 Seconds.
Do Anyone know how to do this?  A new entire in the \client\src\app\device\device-property\device-property.component.ts
file doesnt have any effect for me.

unocelli, 2022-01-24:
Hi,
you have to build if you are looking on localhost:1881
`ng build --prod`

unocelli, 2023-03-19:
I am going to close this as resolved, let us know if you have any issues.

closed 2023-03-19 by unocelli

eri6081, 2025-02-10:
@hongnhan100 Hello, I am facing the exact same issue as you was... How did you solve this problem? For reference, i am using the latest release, and facing this issue: 
As you can see the error tells me, that the object list of the device isn't supported but with every device
[image]