frangoteam/FUXA #142: Any additional event trigger type for "Open Dialog" rather than user "Click"?

studuino, 2021-05-08:
Hi @unocelli, great project.
Is there any update on the way regarding action like "Open Dialog" triggered from sensor value change rather than user click action.
thank you for the great project again.
[image]

unocelli, 2021-05-09:
Hi, Thanks.
Interesting, were you thinking of events such as mouse over?...do you have an example

studuino, 2021-05-12:
Actually I am thinking of to open dialog(e.g. Yes or No question) for operator to answer.
The dialog popup is due to a certain sensor value change.
Sorry if I cannot make it clear.

unocelli, 2021-05-14:
I think that what you mean is more a warning or info notification as dialog form (showed by a certain sensor value change), but this one is independent from controls and shapes event (like click)?

studuino, 2021-05-14:
Yes, this is exactly what I mean. I couldn't express it in proper way.
Thank you for the correction.

unocelli, 2021-05-14:
I like the feature :)
A simple way to add it is to extend the Alarms (as Alarms and Notifications) with the possibility to define the trigger action to show a View as Dialog, the action could be a email too.

studuino, 2021-05-31:
Yes, I am thinking of the output that trigger dialog/card over value change. sorry I am noob to the programming.

studuino, 2021-06-06:
Something like this
[image]

unocelli, 2022-01-30:
added in v1.1.4 
going to close this as resolved. let me know if you have any issues.

closed 2022-01-30 by unocelli