frangoteam/FUXA #149: can you add more module for schedule control

thavornprasertkaew, 2021-05-18:


unocelli, 2021-05-20:
Hi,
Please, can you me explain more in details what you mean?

thavornprasertkaew, 2021-05-21:
I mean  automation set  schedule  when time monday on 02.00 PM - 05.00 PM  auto  OFF to Lamp

unocelli, 2021-05-22:
Yes, I have also planed the feature in form of a scheduler of scripts that run in server side. To make this we need in frontend somethings like a javascript editor as well as an editor to define the scheduler. the feature requires some effort and will take some time.

MatthewReed303, 2022-09-08:
I'm also in need of a scheduler type interface for the UI. Is it possible to use prebuilt node.js scheduler or Node-Red type scheduler. For example:
https://flows.nodered.org/node/node-red-contrib-ui-time-scheduler/in/590bc13ff3a5f005c7d2189bbb563976

unocelli, 2025-12-26:
The scheduler control is available in the latest version